Bodmer/TFT_eSPI #249: getRawTouch always returns true

DeanCording, 2018-11-21:
The getTouchRaw function does not work as advertised.  It always returns true and does not return false as if not pressed.  The code doesn't even read the Z value.

    /***************************************************************************************
    ** Function name:           getTouchRaw
    ** Description:             read raw touch position. Return false if not pressed. 
    ***************************************************************************************/
    uint8_t TFT_eSPI::getTouchRaw(uint16_t *x, uint16_t *y){
      uint16_t tmp;
      CS_H;
    
      spi_begin_touch();
    
      T_CS_L;
      
      // Dummy transfer
      SPI.transfer(0xd0);
      SPI.transfer(0);
      SPI.transfer(0);
    
      // Start bit + YP sample request for x position
      SPI.transfer(0xd0);
      tmp = SPI.transfer(0);
      tmp = tmp <<5;
      tmp |= 0x1f & (SPI.transfer(0)>>3);
    
      *x = tmp;
    
      // Start bit + XP sample request for y position
      SPI.transfer(0x90);
      tmp = SPI.transfer(0);
      tmp = tmp <<5;
      tmp |= 0x1f & (SPI.transfer(0)>>3);
    
      *y = tmp;
    
      T_CS_H;
    
      spi_end_touch();
    
      return true;
    }

Bodmer, 2018-11-21:
It sounds like your are not using an XPT2046 touch controller, it is not wired correctly or there is a bus conflict on the SPI bus.

Please describe your setup, hardware used, pins used and which example sketch you are using.

DeanCording, 2018-11-21:
If you look at the code above from the Touch.cpp library, it only ever returns true - there is no other return path so it can never return anything other than true.