Bodmer/TFT_eSPI #251: TFT_eSPI_Button with Visual Studio Errors

Uwe-HH, 2018-11-21:
Hello Bodmer,

I used your TFT_eSPI Library since some month and the Touch function works excelent. Now I changed the Editor an I use Vusual Studio 17. When I compiele the Sketch I receive the following Message:

Button.cpp: 5:1: error: 'TFT_eSPI_Button' does not name a type
 TFT_eSPI_Button*: TFT_eSPI_Button(void) {

... and some more Errors for the TFT Class.

When I use the Arduino IDE everything works great. I changed the setup.h as you told.

Do you have an idea for this reason?

Thank you

Uwe

Bodmer, 2018-11-22:
This is probably because the Arduino Visual plugin is fairly dumb when deciding what to include, so the button class is not being loaded. Have you tried checking the "Visual Micro>Deep Library Search" menu item.

Uwe-HH, 2018-11-22:
Thank you so much for your quick answer. After hours of searching for a solution I found the mistake 30 minutes ago. During the reinstalling of the Arduino IDE and manually deleting of some folders and then reinstaling some backups, there was a second library TFT_eSPI on the system. In this library I found the old setup.h file.

Maybe Visual Studio uses this reference. I deleted everything a second time and reinstaled some missing or wrong files (e.g. wire.h) and now every thing is OK.

Never change a running system !

Uwe

(Sorry for the mistakes, my English ist not very good)

Bodmer, 2018-11-22:
I am glad you found the solution, it would have been tricky for me to figure out what is wrong.

Since I only know English well for technical matters I do not mind mistakes! I have understood everything you said and that is what is important.

Raising this has prompted me to have a look at Visual Studio again, I have not used it for many years and it is time for me to try it again.

closed 2018-11-22 by Bodmer