Bodmer/TFT_eSPI #252: Touch not working

wb9688, 2018-11-23:
I've a LOLIN32 and a 2.4" 240x320 touch screen with an ILI9341 and a XPT2046. I've connected the following pins on my touch screen to the following pins on my LOLIN32.

```
T_IRQ 15
T_DO 2
T_DIN 4
T_CS 16
T_CLK 17
SDO(MISO) 5
LED 3V3
SCK 18
SDI(MOSI) 23
DC 19
RESET 21
CS 22
GND GND
VCC 3V3
```

In `User_Setup.h` I commented the default pin definitions and set the following pins:

```
#define TFT_MISO 5
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   22
#define TFT_DC   19
#define TFT_RST  21
#define TOUCH_CS 16
```

When I run `Test_Touch_Controller`, I'm getting some random touch coordinates even though I'm not pressing anything:

[image]

`TFT_Pong` works fine though, so the display part is working at least.

How can I get the touch part to work as well? Thanks in advance.

Bodmer, 2018-11-23:
The touch controller is an SPI device that must share the interface with the TFT. Connect as follows:
T_IRQ - no connection
T_DO 5
T_DIN 23
T_CS 16
T_CLK 18

closed 2018-11-24 by Bodmer

wb9688, 2018-11-24:
That works, thanks! It'd be nice if you had more info like this in the
README though.