Bodmer/TFT_eSPI #253: Library not working with ILI9341 SPI display + ESP8266 module

Smashcat, 2018-11-23:
Tested using following pins:
MISO: GPIO12
SCK: GPIO14
MOSI: GPIO13
D/C: GPIO05
RESET: RST
CS: GPIO04

Screen flickers after reset, but does nothing else. This is a bare ESP8266 module. Settings in the User_Setup.h file were set to match.

Bodmer, 2018-11-23:
The ESP8266 has dedicated pins for the SPI interface. The only pins you can allocate are DC, CS, and RST.

Use the pins recommended in the setup file:
// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST, see below)
// Display CS        to NodeMCU pin D8 (or GND, see below)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

closed 2018-11-24 by Bodmer

Smashcat, 2018-12-29:
Hi, thanks, I got it working in the end. The issue seemed to be using the software LCD reset. Once that was removed it was fine!

Bodmer, 2018-12-29:
OK, thanks for letting me know.

Smashcat, 2018-12-29:
Thanks for the library by the way ! I modified it slightly for my game, mostly to get a pointer to the sprite data array for faster access, and it worked well!