Bodmer/TFT_eSPI #254: Wish List :   Example for rending PNG image to TFT Screen.

TridentTD, 2018-11-24:
Sometime picture from internet received by png format.
If there is an example for rending png image,
it's helpful for many usage.

Thank you.

Bodmer, 2018-11-24:
Yes, it is on my to do list.  Decoding a png needs a lot of RAM to hold the image as it is decoded, so an ESP32 with PSRAM will be needed, this is the main reason why an Arduino library is not availabe yet.

closed 2018-11-24 by Bodmer

Andy1968T, 2020-12-27:
Now that TFT_eSPI has taken a good foot hold in LILYGO devices with LCD screens. It would be very useful to be able to render .png files. Else the driver is running the development of useful esp32 LCD based apps to be developed, down a dead end. Anyone achieved this yet? Or found a driver that can that works with an ST7789V  with .png's.This is way beyond my programming skills.

Bodmer, 2020-12-27:
There are two working Arduino sketch examples here:
https://github.com/Bodmer/PNG_TEST_ONLY

The png decoder is part of the sketch, but I plan to create a png decoder library.

VioletGiraffe, 2021-04-01:
Thank you for providing the example! Just want to bump this discussion - would be awesome to have it as a ready-to-use library. ESP32 is a pretty powerful SoC, often used for doing Internet stuff, and in my case I need to download and display small icons (50x50) in PNG format.

Bodmer, 2021-04-01:
Fair enough! I have been distracted with other things. Thanks for the bump!

JHershey69, 2021-06-21:
This is really needed right now... displaying png files from the web on the T-watch, LilyGo T-Display, et all. Just sayin' :)

simkard69, 2021-12-14:
Hello,

Just to know if there is any update on drawing PNG files directly from files (SPIFFS and/or SD card) ?

BTW, Big thanks for the hard work already provided.

Koxx3, 2022-02-19:
> Hello,
> 
> Just to know if there is any update on drawing PNG files directly from files (SPIFFS and/or SD card) ?
> 
> BTW, Big thanks for the hard work already provided.

+1

Bodmer, 2022-02-19:
Examples added here:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/PNG%20Images

Koxx3, 2022-02-20:
> Examples added here: https://github.com/Bodmer/TFT_eSPI/tree/master/examples/PNG%20Images

thanks a lot !
forgot png needed some much ram 😅
40-45K to decode.

Bodmer, 2022-02-20:
The pngle tool is a bit more economical on memory for PNG files, but a bit slower.  I created [some sketches here](https://github.com/Bodmer/PNG_TEST_ONLY) that use it.

Koxx3, 2022-02-20:
yes. i tested your sketches already. i am short in ram, i'll stick with gif 😅

ruebarb, 2022-03-11:
How do you use transparency with the png? For me that is the advantage of using png vs jpg for graphics.

Thanks

I'm working with Flash_PNG

How do I use your function, where do I get the **uint16_t transparent** value?

**TFT_eSPI.h**
void     pushImage(int32_t x, int32_t y, int32_t w, int32_t h, const uint16_t *data, **uint16_t transparent**);

**PNGDec.cpp**
uint8_t PNG::getAlphaMask(PNGDRAW *pDraw, **uint8_t *pMask**, uint8_t ucThreshold)

In the png_transparency example

uint8_t ucMask[40];
png.getAlphaMask(pDraw, ucMask, 255)

juper, 2023-03-14:
co to z apng jak nie ma przexroczystości

Bodmer, 2023-03-14:
See example here:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/PNG%20Images/Flash_transparent_PNG

tnetwork96, 2023-06-17:
> There are two working Arduino sketch examples here: https://github.com/Bodmer/PNG_TEST_ONLY
> 
> The png decoder is part of the sketch, but I plan to create a png decoder library.

I have tried running TFT 80x160 using your examples, but I have failed