Bodmer/TFT_eSPI #255: Add Scrolling for 1-bit sprites

Redstoned, 2018-11-24:
Hi,
I just wanted to scroll an 1-bit sprite but as it seem the library doesn't support it. Could You please implement it?

Bodmer, 2018-11-24:
Adding scrolling for 1 bit Sprites is trickier than for 8 or 16 bit Sprites because in those cases memcpy could be used.

Which direction d you wish to scroll in?

Redstoned, 2018-11-24:
All directions would be best. Are some directions easier than other to implement?

Bodmer, 2018-11-24:
I had a look and scrolling was easy if I use pixel be pixel copying. So I have added scrolling for 1bpp Sprites and added a new example: Sprite_scroll_1bit.

1bpp Sprites are unique in that the coordinate frame can be rotated, so the example shows how this is done too.

closed 2018-11-24 by Bodmer