Bodmer/TFT_eSPI #257: How to screen-capture by processing?

TridentTD, 2018-11-25:
When I upload Arduino's TFT_Screen_Capture.ino  to my TFT+ESP32
and try  Processing's Screenshot_client.pde example 
(  ..\libraries\TFT_eSP\Tools\Screenshot_client\Screenshot_client.pde )
But I can't receive any image.

How to get screen capture image by this processing ?
Thank you.

Bodmer, 2018-11-25:
Run the [example here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/TFT_Screen_Capture), copy the processing sketch out of the examples tab and use that.  If the TFT does not support reading you will get black or white screens transferred. Make sure the processingsketch is using the right serial port, instructions are in the header.

TridentTD, 2018-11-25:
I close all other COM port and select right serial port.
But when I run the processing sketch , it return "no response" as this picture.

https://imgur.com/1RHj9T2

Bodmer, 2018-11-25:
In that case the problem is in the Arduino sketch.

Make sure screenServer(); is being called in your sketch and the screenServer tab is attached to your sketch.

Bodmer, 2018-11-25:
I see from the screen image that you are not using the processing sketch from the example above. You MUST use [this example](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/TFT_Screen_Capture) for testing and you MUST use the copy of the processing sketch (version 0.08) copied from the tab that generic example.  The reason for this is that the ESP32 board support core does not support serial.end().

Bodmer, 2018-11-25:
I have tidied up the library to remove an ESP8266 only version that I suspect has caused issues with the ESP32.  All should be good now.

closed 2018-11-25 by Bodmer

TridentTD, 2018-11-26:
Thank you. The processing can capture now.

ArnieO, 2020-09-11:
Not working for me. 
Running latest version Processing (3.5.4), and proceeding according to instructions.
Error message: 
[image]
All stored images are black.

Please advise.

Koxx3, 2022-03-14:
how do you use the serialScreenServer ? 
which tool do you use to capture the output and put it in a file ?
I don't see the PNG header in it is it normal ?

mm108, 2022-08-23:
> 

Same situation here. I tried looking at all the parameters and tried all different settings I could - both in the capture part on ESP32 as well as the processing arduino sketch. I tried  changing the file location ... different formats etc. The file does not get created and I get a black screen in the TFT image viewer

```
Ready: requesting image *********
................................................................ [ 20% ]
................................................................ [ 40% ]
................................................................ [ 60% ]
............................................................... [ 80% ]
................................................................ [ 100% ]
Image fetch time = 3.335 s

Saving image as "screenshot_2022_8_24_0_29_1.jpg"
Saved image count = 1
```
I have tried all the formats. Have you had any luck with getting it to work @ArnieO ?

[image]

ArnieO, 2022-08-23:
> Have you had any luck with getting it to work @ArnieO ?
No, I never got it working.

mm108, 2022-08-23:
> > Have you had any luck with getting it to work @ArnieO ?
> 
> No, I never got it working.

Oh ok. I think it probably does not work with all chips. I'll try another display to see how it goes. Thanks.

mm108, 2022-08-23:
A small problem on the capturing side. On the PC (TFT image viewer), some colors seem to get changed. For example Yellow appears as Cyan and Cyan appears as yellow! :) Red, blue seems to be fine.