Bodmer/TFT_eSPI #258: ESP32 ILI9341 with SPI bus can't work correctly

imcort, 2018-11-25:
I found the release 1.3.3 and 1.3.2 can't display anything but zebra-stripe on the screen when using the ili9341 spi with esp32.
the release 1.3.0 can work normally.

Bodmer, 2018-11-25:
Thanks for reporting this I made some changes that broke things and tried to quickly fix them! I will look at this later today.

Bodmer, 2018-11-25:
I have checked the 1.3.3 release and it works fine with ESP32 and SPI ILI9341.

I am not sure what happened to your copy of 1.3.3 but I have deleted that release to force the IDE library manager to jump to the next release.

It will take up to 24hrs for the IDE to spot the new release. In the meantime a zip file can be downloaded. Make sure you update the setup for your board.

Report back if you still have problems.

I am going to delete the 1.3.0 release so make a copy of yours before testing 1.3.4.

closed 2018-11-25 by Bodmer