Bodmer/TFT_eSPI #260: ILI9488 not working in v1.3.4

sv1eia, 2018-11-26:
Hi,

The v1.1.3, is the latest version were both ILI9341 and the ILI9488 are working ok. After that in all versions up to v1.3.4 there are issues.

Simply reverting back to v1.1.3 do not work due to caching. It needs to delete the TFT_eSPI library folder and install fresh.
The following tests are clean install from scratch to specific version.
Tested v1.1.3, ILI9341 is working and ILI9488 working
Tested v1.3.0, ILI9341 is working and ILI9488 is not working
Tested v1.3.2, ILI9341 is not working and ILI9488 is working
Tested v1.3.4, ILI9341 is working and ILI9488 is not working

Hope this situation gets stable soon..

sv1eia, 2018-11-26:
Forgot to add, with ESP32.

sv1eia, 2018-11-26:
Looks there is an easy patch to 1.3.4
This line here
https://github.com/Bodmer/TFT_eSPI/blob/beda811a3aed4bcaa6a53a98716fd28c80725179/TFT_eSPI.h#L231
should be changed from
`  #ifdef ESP32_PARALLEL`
to
`  #if defined (ESP32_PARALLEL) || defined (ILI9488_DRIVER)`
and then both ILI9341 & ILI9488 do work ok in 1.3.4

Bodmer, 2018-11-26:
Are you using SPI on both displays? Or is one display on an parallel 8 bit bus?

sv1eia, 2018-11-26:
Both are SPI.

https://www.aliexpress.com/item/3-5-inch-480-320-SPI-Serial-TFT-LCD-Module-Display-Screen-with-Touch-Panel-Driver/32908809356.html?spm=a2g0s.9042311.0.0.27424c4drsn8LA

https://www.aliexpress.com/item/free-shipping-2-2-INCH-TFT-SPI-serial-LCD-resolution-240-320-2-2inch-LCD-for/32913886208.html?spm=a2g0s.9042311.0.0.27424c4drsn8LA

Now with the change I mention above, both work ok again in the patched 1.3.4.

Bodmer, 2018-11-26:
Thanks, I forgot the ILI9488 is different to others in SPI mode only.

Now corrected in 1.3.6

closed 2018-11-26 by Bodmer