Bodmer/TFT_eSPI #261: Latest library versions after 1.1.3 do not work with RPi display with the ESP32

Bodmer, 2018-11-28:
The library control signal timings have been shortened for the ESP32 to improve graphics performance. This is not compatible with the RPi display serial shift register circuit design, the RPi display needs extra delays.

Use version 1.1.4 until this is corrected.

kostuch, 2018-11-28:
Wow... I lost three days and some hair with my head to find out why it doesn't work with esp32 and without any problem on esp8266... And there is just a driver problem.
Look forward for bug fix.
Anyway. I'm bit disappointed with performance of RPI ILI9486 display. I have to decrease SPI clk to 14MHz (on esp8266) to make it work without artifacts. This is cheap China no-name, but I have also no-name generic ILI9341 which is MUCH faster.

Bodmer, 2018-11-28:
Bug fixed a few minutes ago, use 1.3.7 from the repository it will be availble from the Arduino IDE library manager in a few hours when the server check the repository.

I found the same, I rarely use the RPi display anymore but at the time it was the only big 320x480 SPI compatible screen I could find. Now there are low cost 480x320 ILI948x displays about that are quite cheap and are much faster.

Bodmer, 2018-11-28:
Fixed

closed 2018-11-28 by Bodmer