Bodmer/TFT_eSPI #264: ESP8266 in overlap SPI mode is not initializing slow displays

kostuch, 2018-12-01:
I tried to setup ILI9486 (RPi version display) with ESP8266 configured to use SPI in overlap mode.
Apparently clock is the same (ignore settings from .h setup files) when communicate with cpu flash and with display. 40/80MHz is far to much to properly drive slow displays.
Maybe there is a setting in registers to slowdown SPI clock during communication with other device?
If not, then only fast enough devices will work in SPI overlap mode.

Bodmer, 2018-12-01:
Can you try turning on transaction support by uncommenting the last line in the setup.

This should switch the SPI clock rate to a value you choose in the setup file when the display is being accessed. It will be switched back to a high frequency automatically.

If this works then let me know and I will change the library to always turn on transaction support when overlap mode is chosen.

kostuch, 2018-12-01:
Thank you for the hint.
I recompiled with #define SUPPORT_TRANSACTIONS but it didn't help.

Bodmer, 2018-12-02:
I have had a tink about this further and realised I have used overlap mode with slower displays but not the RPi display.

I have connected up an oscilloscope to the CLK pin and can see that during SPI transfers to the TFT display the clock rate does indeed change to the slower speed set in the setup file even if transaction support is not switched on.

The problem with the RPi display is that it needs a slower chip select strobe and more importantly expects all transfers to be in 16 bits. In overlap mode the FLASH controller interrupts at 8 bit boundaries so this means the RPi display does not always get the 16 bit transfers it must have.

Therefore unfortunately because of the 16 bit special need of the RPi display it cannot be used in SPI mode, other slow displays can.  I will add a note to this effect in the setup file.

Thanks for bringing this to my attention.

kostuch, 2018-12-02:
Thank you for clarification. Unfortunately I don't have good enough oscilloscope to watch this effect.
Apparently RPi displays are fine for RPi :)
Appreciate your work.