Bodmer/TFT_eSPI #265: TFT_eSPI ILI9341 PROBLEMS

electromc2, 2018-12-02:
I cant get  your lib to work on ILI9341 with a esp32 board. 
The Display is https://www.banggood.com/2_8-Inch-ILI9341-240x320-SPI-TFT-LCD-Display-Touch-Panel-SPI-Serial-Port-Module-p-1206782.html?rmmds=search&cur_warehouse=CN

The User_setup.h details are:
#define ILI9341_DRIVER#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5// Chip select control pin
#define TFT_DC    2 // Data Command control pin
#define TFT_RST   4 // 
 
The adafruit_ILI9341  lib works ok, so the display is good. Also i have tried it on an esp8266 with your library with no problems.
Any ideas
Terry

Bodmer, 2018-12-02:
This is strange. I have an identical display and it is working fine.

Run the "Read_User_Setup" diagnostic example and check the reported configuration is correct, this verifies the compiler is picking up your configuration correctly. If this is OK then let me know and I will investigate further.

If you have the touch controller pins wired up as well as the TFT then define the chip select pin in the setup file to avoid bus conflicts. Check you are using the latest version of TFT_eSPI.

Bodmer, 2018-12-08:
Closing, if you are still having problems then let me know.

closed 2018-12-08 by Bodmer