Bodmer/TFT_eSPI #266: name of the examples gives error in IDE

lesept777, 2018-12-02:
The name of the examples 1_bit_Sprite_demo and 1_bit_Yin_Yang are not consistent with the rules of the Arduino IDE, as they shouldn't begin with a number. This gives an error messagebox each time the IDE is launched. Can you change them to:

- One_bit_Sprite_demo
- One_bit_Yin_Yang

per1234, 2018-12-02:
These error messages will occur when using Arduino IDE 1.8.3 and older. The restriction was removed in Arduino IDE 1.8.4: https://github.com/arduino/Arduino/commit/4545283ae773b6351d089e37b6d4080905ea8fd5

closed 2018-12-02 by lesept777

per1234, 2018-12-02:
@lesept777 I wasn't implying the issue was invalid, but only describing what was necessary to reproduce the problem.

I certainly recommend that people use the current IDE versions but sometimes bugs have been introduced in the IDE that prevent people from updating, or else they are comfortable with the old IDE version and don't want to risk an update. For these reasons, many people are using old IDE versions. I think it's a good idea to provide a reasonable amount of backwards compatibility when possible (maybe back to 1.6.5).

Bodmer, 2018-12-02:
Thanks for reporting this, I did not know about that restriction as the sketch was created under a later version.

I will rename at the next release which will happen in the next few days.

Bodmer, 2019-01-06:
Later than expected, but this update has now been done. Thanks.