Bodmer/TFT_eSPI #268: SPI Overlap mode not working (ILI9488 & ILI9341)

wvmarle, 2018-12-05:
I'm trying to get my displays to work in overlap mode but no luck. One ILI9341 240x320 display, one ILI9488 320x480 display.
Both work fine with regular SPI.
Default SPI bus speed (27 MHz).
Output from user_setup looks exactly as expected, display is wired as per instructions (and DC to D5, RST to D4). Same result for RST connected to 3V3.

[code]
Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.25 V
Transactions = No 
Interface    = SPI 
SPI overlap  = Yes 

Display driver = 9488
Display width  = 320 
Display height = 480 

Overlap selected, following pins MUST be used:
MOSI     = SD1 (GPIO 8)
MISO     = SD0 (GPIO 7)
SCK      = CLK (GPIO 6)
TFT_CS   = D3  (GPIO 0)

TFT_DC and TFT_RST pins can be user defined
TFT_DC   = D5 (GPIO 14)
TFT_RST  = D4 (GPIO 2)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
[/code]

Any ideas? It'd be awesome to free up those SPI pins! In between Serial, I2C and SPI I'm pretty much full.

Bodmer, 2018-12-05:
The overlap mode should work fine with the ILI9341. So try with that display.

It appears that some displays are not compatible with the overlap mode because they do not tolerate the interruptions in communication caused by sharing the bus with the program FLASH.

I implemented overlap mode for a library user and tested it with ePaper displays and the ILI9341 only. I did not anticipate this potential issue with other display types.

I will investigate further but in the meantime try the ILI9341 display again and verify the settings as you have with the ILI9488.  I have made some significant changes to the ESP32 code that perhaps has impacted on the working of the ESP8266 so I will check this as well.

wvmarle, 2018-12-05:
Same, same. Screen remains white; nothing drawn also for the ILI9341. Below the output. Both boards have the exact same pinout so same wiring, just plugged in the other one (solderless breadboard).

Thanks for the work writing & maintaining this library! It's very useful. The normal SPI mode works perfectly.

Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.25 V
Transactions = No 
Interface    = SPI 
SPI overlap  = Yes 

Display driver = 9341
Display width  = 240 
Display height = 320 

Overlap selected, following pins MUST be used:
MOSI     = SD1 (GPIO 8)
MISO     = SD0 (GPIO 7)
SCK      = CLK (GPIO 6)
TFT_CS   = D3  (GPIO 0)

TFT_DC and TFT_RST pins can be user defined
TFT_DC   = D5 (GPIO 14)
TFT_RST  = D4 (GPIO 2)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz

Bodmer, 2018-12-05:
I will setup the displays using overlap mode and test. It looks like some recent updates have introduced a bug.

Bodmer, 2018-12-05:
I have connected up my ILI9341 TFT in overlap mode and it works fine.  However I suspect you have TFT displays that are configured for bi-directional mode and these do not always tri-state the SDO pin when chip select is high.

Try your displays again but DO NOT connect the SDO pin from the TFT to the ESP8266 MISO/SD0 pin.

Let me know if this works for you. If it does then I will update the documentation.

Bodmer, 2018-12-08:
Just realised what the problem is.  [This line in the setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L210) needs to be uncommented. 

I plan to remove the need for this as it is easily missed.

wvmarle, 2018-12-09:
No change. Still doesn't work.
I wonder what effect it would have in the first place as D0 (GPIO16) is not connected in the first place.

I have now uncommented in User_Setup.h:

#define ILI9341_DRIVER
#define TFT_SPI_OVERLAP
#define TFT_CS   PIN_D3
#define TFT_DC   PIN_D5  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)

#define D0_USED_FOR_CS

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000 // Actually sets it to 26.67MHz = 80/3
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

wvmarle, 2018-12-09:
Yes, that one is disconnected. Tried that before. Tried only the ILI9341 2.8" 320x240 display now, as that's the driver that works for you.

I have these connections now (NodeMCU pin - display pin):
S1 - SDI(MOSI)
SK - SCK
D5 - DC
D4 - RESET
D6 - CS
GND - GND
3V - VCC
3V - LED
Using the "TFT_Rainbow_one_lib" example from the library as test.

S0 - SDO(MISO) connection I already removed.

Display remains grey.

BTW I think documentation needs lots of improvements, it took me initially quite a while to find the required wiring (and that the DC pin has to be connected somewhere) in the User_Setup.h file. There should be a lot more in the readme about those things as well. Also which exact defines to enable/disable for these special tricks.

Bodmer, 2018-12-09:
Yes, the documentation is lacking and that is on my todo list but paid for work and other activities tend to take priority.

What version of the Arduino IDE, TFT_eSPI and the ESP8266 board support package are you using?

The NodeMCU pin names used in you last post (S1, SK) are not the same as mine, have you listed the names used on the circuit board? Can you provide a link to the NodeMCU board you have so I can verify the names tie up.

wvmarle, 2018-12-24:
This is the setup I've been using:

Arduino IDE: 1.8.7
NodeMCU: that are indeed the markings as given on the PCB. Link: https://item.taobao.com/item.htm?spm=a1z09.2.0.0.4c292e8d7q4BS8&id=543020479619&_u=o24vmoevc125
TFT_eSP: 1.3.9 (as given in library.properties)

Bodmer, 2018-12-28:
I cannot reproduce the problem. My displays work fine in overlap mode. Can you post pictures of your setup so I can check the wiring is as expected.

Bodmer, 2019-01-04:
Just looked at this again and checked your pin settings compared to mine. In ESP8266 overlap mode the [TFT chip select MUST connect to pin D3](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L134), so this will be the problem in your setup.

wvmarle, 2019-01-04:
Just saw your comment.

Double checked connections: D3 is indeed connected to CS on the board. Wiring is as far as I can tell correct, yet it doesn't seem to work.

[image]
[image]
[image]

wvmarle, 2019-01-04:
I even checked the connections with my multimeter - NodeMCU PCB solder point to display solder point - and all are confirmed connected.

Enabled lines in User_Setup.h:
#define ILI9341_DRIVER
#define TFT_SPI_OVERLAP
#define TFT_CS   PIN_D3
#define TFT_DC   PIN_D5  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
#define D0_USED_FOR_CS
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT
#define SPI_FREQUENCY  27000000 // Actually sets it to 26.67MHz = 80/3
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

Bodmer, 2019-01-05:
Hi,
I have created a new build and wired it identically to yours. The only difference I can see is that my board is a NodeMCU 1.0, I see you have a v3 board, but I do not have one of those.

This is the setup:
```
//#define ILI9488_DRIVER
#define ILI9341_DRIVER

// SD0 = MISO = Do not connect to TFT SDO pin unless TFT tri-states this pin when CS is high
// SD1 = MOSI = TFT pin SDI
// CLK = SCLK = TFT pin SCK

#define TFT_CS   PIN_D3
#define TFT_DC   PIN_D5
#define TFT_RST  PIN_D4

#define TFT_SPI_OVERLAP

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000

#define SPI_TOUCH_FREQUENCY  2500000

```
I uploaded a sketch and it ran fine.
[image]

Unfortunately I cannot do any more to help resolve this. The only difference I can see is that you have a NodeMCU V3.

ZL1CVD, 2019-09-24:
NodeMCU v3.0 board has different pinout. 

I can confirm SPI overlap does indeed work. Pinout as follows:

Board Marking <> ESP12 Pin
S3 <>  GPIO9
S2 <> SCLK
S1 <> MOSI
SC <> CS0
S0 <> MISO
SK <> GPIO10

Thank you for an excellent library BTW!

Bodmer, 2019-09-25:
Thanks for this information, I will add this to the setup notes.

Bodmer, 2019-11-11:
Notes added to setup files.

closed 2019-11-11 by Bodmer

RmkJersey, 2020-02-14:
I have just purchased a Kuman 3.5 inch Tft Touch Screen LCD Display (ILI9486_DRIVER) but I cannot get it to work with any of the drivers ILI948 in your library.  However it does work if I use the ILI9481_DRIVER  with the following pin settings:

Also the tft.readRect function does not appear to work so does the ILI9486 have the ability to read from the TFT screen?

Your library is superb and extremely fast. so keep up the excellent work!

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

#define TFT_RD   23
#define TFT_WR   19 
#define TFT_DC   18 
#define TFT_CS   33
#define TFT_RST  32

#define TFT_D0   12 
#define TFT_D1   13 
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   21
#define TFT_D5   5
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD  
#define LOAD_FONT2  
#define LOAD_FONT4  
#define LOAD_FONT6
#define LOAD_FONT7 
#define LOAD_FONT8
#define LOAD_GFXFF