Bodmer/TFT_eSPI #269: Curiosity Only (Screen Size of Create_Smooth_Font/Create_font)

glown, 2018-12-07:
I typically use the Adafruit gfx library to convert fonts to use in ESP projects.  Lately I've noticed staircasing in the fonts rendered (especially in the number 5) so I ran the Processing sketch Create_font.  It works great!  But I found something that's crazy enough to pique my curiosity.  I apologize for taking your time since I have an easy solution, but the issue nags me.

Here's the thing, a displayed font size for a Create_font vlw file is about 1/2 the size displayed font size for a gfx converted font.  I had been using a gfx 66 pt font, but to get the same size on the screen I need to use a Create_font 132 pt font.  It's so easy to adapt, but it's so crazy I can't stop wondering -- is one of the converters making fonts too large or too small?  

A good and proper response is "just let it go" …. but heck, maybe you know???

Bodmer, 2018-12-07:
Point sizes represent physical dimensions of the font on for example a paper print.  TFT screens have different numbers of pixels per inch so the pixel density of the screen in DPI has to be used to see the right visual size. Adafruit decided on a value of [141 DPI](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/fontconvert/fontconvert.c#L28), the processing environment uses a different value more appropriate for a computer screen.

Bodmer, 2018-12-07:
As far as I can tell the processing environment uses 72 dpi, this is better in my view that the Adafruit approach since 1 point becomes 1 pixel.  The Adafruit a 12pt font occupies 24 pixels height where as a 24pt smooth font is 24 pixels high.  Hence the factor of 2 difference.

Although technically Adafruit have taken the right approach for one display it is only appropriate for that particular display, from the users convenience viewpoint the processing approach is far better since font point to pixels size has a 1:1 correspondence and different displays have different DPI anyway.

closed 2018-12-07 by Bodmer

glown, 2018-12-07:
wow, that all makes sense.  thank you!