Bodmer/TFT_eSPI #270: Create_font suggestion

glown, 2018-12-07:
I needed to make several different fonts sizes so I thought a loop might help.  Here's something I did, not sure if it fits into the spirit of the original code or not, but it made things a lot easier for me:

Original code:
350  void setup() { 
351  
352    size(1000, 800); 

My changes (1. rename setup to setupaux, 2. comment //size (1000, 800), 3. create new setup(), 4. add size(1000, 800), 5. add loop 6. run to create a library of font sizes

350  void setupaux() { 
351  
352    //size(1000, 800);

then at end of file add:

void setup() {
  size(1000, 800);
  for (fontSize = 4; fontSize < 201; fontSize++) {
    setupaux();
  }
}

glown, 2018-12-07:
One last thing … I had trouble successfully running the processing sketch on my Windows 10 machine, but everything worked well on Windows 8.1.   I remember reading about some different font handling in W10, but I don't know enough to be able to say it's a W10 issue or just my instance of W10.

closed 2018-12-07 by glown

Bodmer, 2019-01-06:
I have tried a Windows 10 PC with the latest version of Processing and have not seen any issues.

Thanks for the font loop suggestion, I will try it.

glown, 2019-01-06:
I ran Processing on my main Windows 10 machine last week and it worked for me too. I have no idea what I was doing wrong, but heck, it's working now.

A smoothed font can really make things look good.  Thank!