Bodmer/TFT_eSPI #271: Wrong pin assignment

maltegwde, 2018-12-08:
Hi,
im using the WEMOS D1 Mini and a ILI9341 Display (https://wiki.wemos.cc/products:d1_mini_shields:tft_2.4_shield). 

However i'm unable to get it work. The following sketch using the Adafruit_ILI9341 library works: 

```
#include <SPI.h>
#include <Adafruit_ILI9341.h>

#define TFT_CS D0  //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)
#define TFT_DC D8  //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)
#define TFT_RST -1 //for D1 mini or TFT I2C Connector Shield (V1.1.0 or later)

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);

void setup()
{
  tft.begin();

  tft.fillScreen(ILI9341_RED);
  delay(1000);
  tft.fillScreen(ILI9341_GREEN);
  delay(1000);
  tft.fillScreen(ILI9341_BLUE);
  delay(1000);
}
void loop() {
}
```

But TFT_eSPI/examples/320 x 240/TFT_graphicstest_one_lib/TFT_graphicstest_one_lib.ino with the following User_Setup.h pin defintions doesnt work: 

```
#define TFT_CS   PIN_D0  // Chip select control pin D8
#define TFT_DC   PIN_D8  // Data Command control pin
#define TFT_RST  -1  // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V
```

Any Idea what I'm doing wrong?

maltegwde, 2018-12-08:
Thanks, that fixed it!

closed 2018-12-08 by maltegwde

Bodmer, 2019-01-06:
Library has now been updated to remove the need for #define D0_USED_FOR_DC/CS
(Note: This was only needed for ESP8266 if pin D0 (GPIO 16) was used for the DC or CS signals.)