Bodmer/TFT_eSPI #273: 25% offset on tft clock

jmartens2, 2018-12-22:
hi i always loved your work. im trying to get the tft_clock to move over 25% of the screen. 
ive been editing to get my st7789 to get set up but i cant figure out how to offset.
i see the clock on 75% of the screen the other 25 is pixelated dots

Bodmer, 2018-12-23:
Invert the colours has two possible meanings to me..

To swap blue and red use the [setup file options](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L38).

To invert all colour bits use:
tft.invertDisplay(true);  // or false for no inversion

jmartens2, 2018-12-23:
Thank you. I’ve always enjoyed your work! Thank you for all your hard work and time. I’d like to buy you a beer someday!

Bodmer, 2018-12-24:
Thanks for the feedback. If you have any more questions then I will try to help.

closed 2018-12-24 by Bodmer