Bodmer/TFT_eSPI #274: ESP32 WROVER Kit, call to setTextSize() hangs

rleinfellner, 2018-12-24:
I am using the TFT_eSPI library with the ESP32 WROVER kit, its all configured and I can print text ok, however when I use setTextSize() it just hangs. Any ideas would be appreciated.

	TFT_eSPI tft = TFT_eSPI();
	tft.init();

	pinMode(LCD_BACKLIGHT, OUTPUT);
	digitalWrite(LCD_BACKLIGHT, false);
	tft.fillScreen(TFT_BLACK);
	tft.setTextColor(TFT_WHITE);
	tft.setTextFont(4);
//	tft.setTextSize(1); //If I try to set a font size it hangs
	tft.drawString("Hello", 0, 0);

Bodmer, 2018-12-24:
Thanks for reporting this, it is likely that I have introduced a bug at some point. I will run some tests and see if I can reproduce the problem.

Bodmer, 2018-12-24:
I have run this on my ESP32 setup (which is not a Wrover kit) and it runs fine:
```

#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

void setup(void) {
  Serial.begin(115200);

  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE);

  tft.setTextFont(4);
  tft.setTextSize(1);
  tft.drawString("Hello", 0, 0);
}

void loop() {
  while(1)
  {
    Serial.println("Hello");
    delay(1000);
    yield();
  }
}
```
 so I am not sure what the problem is.

Which version of the Arduino IDE are you using and which ESP32 board support package version?

rleinfellner, 2018-12-27:
Many thanks for testing this for me.  I went back to my code there was an update to the TFTeSPI library, I installed this and now it seems to work fine. 

Also worth noting that simply inserting a SDCard into the WROVER kit will stop the LCD working(boy that was confusing), I think GPIO21 is used on both, so if anyone gets a white screen, that could be the issue.

For future reference the pins for the WROVER Kit V4 are

#define TFT_MISO 25
#define TFT_MOSI 23
#define TFT_SCLK 19
#define TFT_CS    22  // Chip select control pin
#define TFT_DC    21  // Data Command control pin
#define TFT_RST   18  // Reset pin (could connect to RST pin)
#define TFT_BL   5  // NB, Low is on

closed 2018-12-27 by rleinfellner

Bodmer, 2018-12-27:
Thanks for the information.

I had a look at the schematic of the WROVER Kit V4 and the TF card detect pin is connected to GPIO21, this means when a TF card is inserted a mechanical card detect switch shorts GPIO21 to ground. This is a very poor design and really there ought to be say a 1 K series resistor to prevent shorting out GPIO21. The camera interface also uses that GPIO21 signal as a clock.