Bodmer/TFT_eSPI #275: ESP32 & TFT_eSPI::textWidth(char const*, int)

zoli2750, 2018-12-25:
Hi Bodmer, 
I have encountered a "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled." issue on my Widora Air clone.
I'm on Arduino IDE 1.8.5, Win10, Board library installed today (version 1.0.0) from github/espressif/arduino-esp32
 
TFT and DS3231 not connected to the board, just the USB for the upload.
PIN settings for the TFT is the standard in the user setup file.

Excpetion Decoder says:
PC: 0x400d671e: TFT_eSPI::textWidth(char const*, int) at C:\Users\zoli\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 2271
EXCVADDR: 0x00000000

Decoding stack results
0x400d671e: TFT_eSPI::textWidth(char const*, int) at C:\Users\zoli\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 2271
0x400d6745: TFT_eSPI::drawString(char const*, int, int, int) at C:\Users\zoli\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 4299
0x400d1bea: loop() at C:\Users\zoli\Documents\Arduino\BasicOTA_ESP32zoli/BasicOTA_ESP32zoli.ino line 86
0x4012cafa: loopTask(void*) at C:\Users\zoli\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\main.cpp line 17

The affected line in my sketch:
 tft.drawString(rtc.dateFormat("Y-m-d H:i:s", dt), 60, 20, 1);
the same when the date format is shorter: 
 tft.drawString(rtc.dateFormat("i:s", dt), 230, 80, 2);

Using the library:
https://github.com/jarzebski/Arduino-DS3231
the problematic function in this library:
https://github.com/jarzebski/Arduino-DS3231/blob/ea18fe451ac82b524f03a94e5ac27f172c00a1d9/DS3231.cpp#L171

Uncommenting the lines containing the dateformat function there is no Guru panic issue.

I assume the issue is that either the TFT_eSPI or the ESP32 core are have conflicting calculations for the length or for the width of a textstring when it is produced by a somewhat complex function.

On Nodemcu 1.0 12E there is no such problem, TFT displays the string : " rtc.dateFormat("Y-m-d H:i:s", dt) 
" correctly without any issue. 
You can surely assign it to the ESP32 support if there is no TFT_eSPI implication.

Bodmer, 2018-12-26:
The problem is in the DS3231 library.

Within the dateFormat libraries dateFormat function the author has created a buffer to contain the result, this buffer only exists in that function and is released upon exit of the function. The function therefore returns a pointer to a buffer that no longer exists because the function ends! The ESP32 is very intolerant of coding errors like. With other processors you can "get lucky" and the problem is a hidden latent bug.

You need to find a DS3231 library that is actively updated.

Bodmer, 2018-12-26:
If you [look here, you will find a zip copy](https://github.com/jarzebski/Arduino-DS3231/issues/22) in one of the posts that fixes the problem, it makes the buffer persist so that it is always resident in RAM. Not so good a solution for an UNO with limited memory but it should not be such a big deal for an ESP32.

closed 2018-12-26 by Bodmer

zoli2750, 2018-12-26:
Dear Bodmer, 
I thank You for your quick response and time,
and I would like to express my real sorry for forcing You to search for similar issues what I could have done by myself.
Next time I will do the research before reporting an issue.
Thank You.
:)

Bodmer, 2018-12-26:
No problem! Happy to help.