Bodmer/TFT_eSPI #277: HSPI mapping not working?

AlexFerreira1984, 2018-12-28:
Hello!
I am using a custom board, with a WROOM_32U ESP32 module. My LCD (ILI9341 - SPI wired) is connected to HSPI Pins (MISO - 12, MOSI - 13, SCLK - 14) and CS - 25, DC - 32, RST - 4. 
Although I made the due changes both in User_Setup and User_Setup_Select (Driver, Pinout, etc), the LCD does not work. 
Tried the same setup with Adafruit_GFX and Adafruit_ILI9341, with "sethWcS(true) and it works great. 
Am I missing some configuration to force SPI PIN mapping?

Tks!

Bodmer, 2018-12-29:
The TFT_eSPI library uses the VSPI interface which natively maps to pins 18, 19, 23 and 5 for hwrdware CS. The HSPI port maps to 12, 13, 14 and 15 for hardware CS. If the SPI pins are mapped to non-native pins then they are connected through a GPIO multiplexer which affects the interface speed.

I will need to run some tests to see if this is the issue.

The reason I use the VSPI port is partly arbitrary.  Low cost boards typically wire the TFT to those pins too. It would be easy to add an option to use HSPI if that is needed.

Bodmer, 2019-01-02:
The sethWcs currently requires the native pin 5 to be used, I have tested this and it works OK. To invoke this the TFT_CS must not be defined in the setup file at all.

The library does not provide the facility to remap pin 5 to pin 25, so it will not work with your setup. This will also be the case if I implement the HSPI interface which uses pin 15. If you define TFT_CS as pin 25 in the setup it will be driven in software. In practice the performance improvement with hardware CS is tiny (<1%) on average and would not be noticed.

Bodmer, 2019-01-02:
Interesting, I have run tests and hardware driven CS is slower than software driven CS. For fill screen it is 0.3% slower and for a drawPixel test that filled the screen it was 2% slower.

I suspect this is because the hardware drives the CS line high whenever the SPI output buffer empties and this introduces extra delays.

Bodmer, 2019-01-02:
I have tried an identical pin set to yours and my ILI9341 display works fine with no other changes to the library.

Check your setup is correct and being picked up by the compiler, use the /Test and diagnostics/Read_User_Setup/ example to check.

If you have other SPI devices connected to the same bus then make sure the chip selects to those devices are set to the high state before intialising the TFT. I am not sure what else I can do to help.

This is the setup I used, using the pin numbers from your first post:
```
#define ILI9341_DRIVER

#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   25
#define TFT_DC   32
#define TFT_RST   4

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000

#define SPI_TOUCH_FREQUENCY  2500000
```

AlexFerreira1984, 2019-01-03:
Good morning!
Will do the test, by the end of the day I'll post the result!
Thank you very much!

AlexFerreira1984, 2019-01-03:
It checks out, I have another SPI device and it was LOW on boot (before initializing the LCD). 

Thank you and sorry for the inconvenience.

closed 2019-01-03 by AlexFerreira1984

Bodmer, 2019-01-04:
Glad to hear it is fixed.