Bodmer/TFT_eSPI #278: pushImage: Different behaviours of _swapBytes

Alonsog2, 2018-12-28:
I have had some problems while displaying images from two different sources: some of them from PROGMEM arrays and others from a dynamic pointer (uint16_t * ptr) which I point to different images along the sketch.
Finally, looking inside the library, I've notice that, though the two functions are apparently similars, the processing of the _swapBytes value is in fact the oposite:

/***************************************************************************************
** Function name:           pushImage - for FLASH (PROGMEM) stored images
** Description:             plot 16 bit image
***************************************************************************************/
void TFT_eSPI::pushImage(int32_t x, int32_t y, uint32_t w, uint32_t h, const uint16_t *data)
....
pushColors(pix_buffer, 64, !_swapBytes);
....

AND in the other similar function:

/***************************************************************************************
** Function name:           pushImage
** Description:             plot 16 bit colour sprite or image onto TFT
***************************************************************************************/
void TFT_eSPI::pushImage(int32_t x, int32_t y, uint32_t w, uint32_t h, uint16_t *data)
....
pushColors(data, dw, _swapBytes);
....

What I do in my sketch is to change the value of swapBytes using setSwapBytes(), but I think that it is inconsistent to have that different behaviour depending on what function you use. Don't you think so?
Thanks

Bodmer, 2018-12-29:
Yes, this is due to the way the library evolved whilst supporting legacy sketches ported from other processors with different endianess...

Rationalising the endianess woul be a good idea so maybe it is time to meke it consistent.

Bodmer, 2019-01-06:
The functions now have the same behavior in latest version 1.3.11.

The Flash bitmap example has been moved to the Generic folder and updated to compensate for the change.

closed 2019-01-06 by Bodmer