Bodmer/TFT_eSPI #279: Reduce cpu time used to handle touchscreen

mstrens, 2018-12-31:
First thanks a lot for this great library.
I am not reporting a bug but I have a question.
 
I have a project based on ESP32 with a 3.2 touchscreen having an XPT2046.
It should be written with Arduino IDE.

I plan to use your library and I expect to read the touchscreen in the main loop.
I would like to reduce as much as possible the time spent in getTouch() in order to give more cpu time to other critical functions.
I can't yet measure the delay requested by the function getTouch() because I am still expecting the delivery my touchscreen.

I looked at the code in Touch.cpp and if I understand the code correctly, I expect that this funtion can sometime request a lot of time to execute.
- First, getTouch() calls 5 X the function validTouch() but uses only the last valid result.
Would it not be more efficient to stop calling validTouch as soon as we get one valid result?
- Second, if Irq_Pin from XPT2046 is connected to a cpu pin configured as input (even without interrupt), I presume we could skip the 5 calls to validTouch() and return false immediately. This would save a lot of SPI access.
- Third, validTouch() begins with this code 

 // Wait until pressure stops increasing
  uint16_t z1 = 1;
  uint16_t z2 = 0;
while (z1 > z2)
  {
    z2 = z1;
    z1 = getTouchRawZ();
    delay(1);
  }  

It means that if the user press slowly and more and more on the screen, the "while" loop can take lot of time before ending.
Is it really needed to wait until pressure stops increasing?

So, in general, would it not be more efficient to:
- take care of level of irqPin (when connected) to detect that screen is not pressed.
- if irqPin is LOW (screen pressed) or not implemented, call getTouchRawZ() and check if it is bigger than  threshold. If lower, than screen is not pressed (so exit).
- else ( if upper than threshold), read e.g. 3 time X and Y and take the 2 positions that are the closiest form each other.

Thanks in advance for your advice.

mstrens, 2019-01-01:
Thanks for your explanation and your comments.
25 ms is already quite long and I would like to reduce the blocking time in the main loop.
Perhaps I can put your touch code in another task because ESP32 is running freetos and the code contains some delay() functions that would allow freetos to switch to another task (e.g. the main loop) when there is a delay().. 

Without implementing a separate task, I already prepared some changes that I can test when I got my screen.
Here the code I have currently prepared (partly based on another lib that I founded):

static int16_t besttwoavg( int16_t x , int16_t y , int16_t z ) {
  int16_t da, db, dc;
  int16_t reta = 0;
  if ( x > y ) da = x - y; else da = y - x;
  if ( x > z ) db = x - z; else db = z - x;
  if ( z > y ) dc = z - y; else dc = y - z;

  if ( da <= db && da <= dc ) reta = (x + y) >> 1;
  else if ( db <= da && db <= dc ) reta = (x + z) >> 1;
  else reta = (y + z) >> 1;   //    else if ( dc <= da && dc <= db ) reta = (x + y) >> 1;

  return (reta);
}

boolean TFT_eSPI::getTouchRawOptimised(uint16_t *x, uint16_t *y, uint16_t threshold) {   // to do : test if this function can be integrated in getTouch() to increase performance
 int16_t data[6];
	CS_H;

  spi_begin_touch();

  T_CS_L;

  // Z sample request
  int16_t tz = 0xFFF;             
  SPI.transfer(0xb1);                // Z1
  tz += SPI.transfer16(0xc1) >> 3;   // Z2
  tz -= SPI.transfer16(0x91) >> 3;   // X
  if (tz >= threshold) {
    SPI.transfer16(0x91 /* X */);  // dummy X measure, 1st is always noisy
    data[0] = SPI.transfer16(0xD1 /* Y */) >> 3;
    data[1] = SPI.transfer16(0x91 /* X */) >> 3; // make 3 x-y measurements
    data[2] = SPI.transfer16(0xD1 /* Y */) >> 3;
    data[3] = SPI.transfer16(0x91 /* X */) >> 3;
  }
  else {
	  data[0] = data[1] = data[2] = data[3] = 0; // Compiler warns these values may be used unset on early exit.
  }
  data[4] = SPI.transfer16(0xD0 /* Y */) >> 3;  // Last Y touch power down
  data[5] = SPI.transfer16(0) >> 3;

  T_CS_H;

  spi_end_touch();

  //Serial.printf("z=%d  ::  z1=%d,  z2=%d  ", z, z1, z2);
//  if (z < 0) z = 0;
  if (tz < threshold) { // if ( !touched ) {
//    zraw = 0;
    return false;
  }
//  zraw = z;
  
  // Average pair with least distance between each measured x then y
  //Serial.printf("    z1=%d,z2=%d  ", z1, z2); Serial.printf("p=%d,  %d,%d  %d,%d  %d,%d", zraw, data[0], data[1], data[2], data[3], data[4], data[5]);
  * x = besttwoavg( data[0], data[2], data[4] );
  * y = besttwoavg( data[1], data[3], data[5] );
	return true ;
}

 
/***************************************************************************************
** Function name:           getTouch
** Description:             read callibrated position. Return false if not pressed. 
***************************************************************************************/
#define MICHEL_STRENS_VARIANT

uint8_t TFT_eSPI::getTouch(uint16_t *x, uint16_t *y, uint16_t threshold){
  uint16_t x_tmp, y_tmp, x_tmp2, y_tmp2, xx, yy;
  if (threshold<20) threshold = 20;
#ifdef  MICHEL_STRENS_VARIANT
   #ifdef TS_IRQ_PIN
		if( digitalRead(TS_IRQ_PIN) ) return false ; // no touch if IRQ_PIN is HIGH
	#endif
		if ( ! getTouchRawOptimised(&x_tmp,&y_tmp ,threshold ) ) return false ; // no touch if pressure is to lower  
		*x = x_tmp;
		*y = y_tmp;
     uint8_t valid = 1 ;
#else
  if (_pressTime > millis()) threshold=20;

  uint8_t n = 5;
  uint8_t valid = 0;
  while (n--)
  {
    if (validTouch(&x_tmp, &y_tmp, threshold)) valid++;;
  }

  if (valid<1) { _pressTime = 0; return false; }
  
  _pressTime = millis() + 50;
#endif
  if(!touchCalibration_rotate){
    xx=(x_tmp-touchCalibration_x0)*_width/touchCalibration_x1;
    yy=(y_tmp-touchCalibration_y0)*_height/touchCalibration_y1;
    if(touchCalibration_invert_x)
      xx = _width - xx;
    if(touchCalibration_invert_y)
      yy = _height - yy;
  } else {
    xx=(y_tmp-touchCalibration_x0)*_width/touchCalibration_x1;
    yy=(x_tmp-touchCalibration_y0)*_height/touchCalibration_y1;
    if(touchCalibration_invert_x)
      xx = _width - xx;
    if(touchCalibration_invert_y)
      yy = _height - yy;
  }

  if (xx >= _width || yy >= _height) return false ; // changed by ms (it was return valid)

  _pressX = xx;
  _pressY = yy;
  *x = _pressX;
  *y = _pressY;
  return valid;
}

Bodmer, 2019-01-02:
I have completed the "TO DO" and added a new function: convertRawXY(uint16_t *x, uint16_t *y);

The PENIRQ line output is also never disabled now.

Simplistically this would work:
```
  uint16_t x = 0, y = 0 ;
  uint16_t z = tft.getTouchRawZ();
  if ( z>200) { //pressed
    tft.getTouchRaw(&x, &y);  // Read raw x and y
    tft.convertRawXY(&x, &y); // Convert to screen coordinates
    tft.fillCircle(x, y, 2, TFT_WHITE);
  }
```
But you will see a lot of spurious x,y values that need filtering. If you run this simple code you will see what I mean. You may need delays to allow for position settle time, this varies significantly between screen suppliers and also with the touch position! The library caters for the worst screen I have and so you may be able to optimise the delays for your particular display. Some experimentation will be needed to get a fast algorithm with zero or tolerable spurious position frequency.  Watch out for spurious x,y when pressure has just been released too.

The existing getTouch() is optimised for the finger touch which is much more prone to spurious coordinates on these cheap resistive screens due to the larger finger contact area.  Using a stylus will give far fewer spurious readings than a finger press but is often inconvenient to use.

One spurious coordinate test I use is to draw an OXO game on the screen with a stylus and draw a circle with a finger.

closed 2019-01-02 by Bodmer

mstrens, 2019-01-06:
I am still waiting for my display.
Still I looked at the texas application note.
As I understand, one of the reason of the noisy data is because the voltage needs some delay to settle when the fet are activated for one reading
They propose 2 solutions but in both, the X must be read several times before reading the Y several times.
I see that in getTouchRaw() you read X only once and then Y only once.
So each time you read X or Y, the voltage is not yet settled.

Perhaps, would you get better results if you made several consecutive reads of X and afterwards of Y with a code like this.

 int16_t data[6];
	CS_H;

  spi_begin_touch();

  T_CS_L;

  // Z sample request
  int16_t tz = 0xFFF;             
  SPI.transfer(0xb1);                // Z1
  tz += SPI.transfer16(0xc1) >> 3;   // Z2
  tz -= SPI.transfer16(0x91) >> 3;   // X
  if (tz >= threshold) {
    SPI.transfer16(0x91 /* X */);  // dummy X measure, 1st is always noisy because it take time to settle
    data[0] = SPI.transfer16(0x91 /* X */) >> 3;
    data[1] = SPI.transfer16(0x91 /* X */) >> 3; // make 3 x-y measurements
	data[2] = SPI.transfer16(0xD1 /* Y */) >> 3; // first Y measure; discarded, 1st is always noisy because it take time to settle
	SPI.transfer16(0xD1 /* Y */) ;
    data[3] = SPI.transfer16(0xD1 /* Y */) >> 3;
  }
  else {
	  data[0] = data[1] = data[2] = data[3] = 0; // Compiler warns these values may be used unset on early exit.
  }
  data[4] = SPI.transfer16(0xD0 /* Y */) >> 3;  // Last Y touch power down
  data[5] = SPI.transfer16(0) >> 3;

  T_CS_H;

  spi_end_touch();

and then using data[] to take 
- the last X Y
- or the average of X and of Y
- or the average of the 2 that are the most similar