Bodmer/TFT_eSPI #280: text is mirrored

jmartens2, 2019-01-01:
hi i have 1.8tft spi 128x160 v1.1 red colored board and my text is mirrored.
im using the st7735 part of the code. everything works just the text is backwards.
i see on internet about changing MADCTL_MX but its greek to me still.
thank you for all your time and energy

Bodmer, 2019-01-02:
Try each option here, except those for different resolution displays.  Recompile and upload and you will probably find one of the options works correctly.

The screen protector tabs are no longer a good indicator of what option to use.

juergs, 2019-01-07:
Choosing case 7 for sainsmart display ST7735 128x160  has done the job for me, enabling the reversed mode:
 case 7:
#ifdef M5STACK
      writedata(TFT_MAD_MX | TFT_MAD_BGR);
#else
      writedata(TFT_MAD_MY | TFT_MAD_MV | TFT_MAD_BGR);
#endif
      _width  = _init_height;
      _height = _init_width;
      break;

jmartens2, 2019-01-07:
Thank you very much

Bodmer, 2019-01-22:
Thanks juergs.

closed 2019-01-22 by Bodmer

Mr-Essid, 2024-04-29:
Thanks @juergs