Bodmer/TFT_eSPI #281: Not compiling; pgmspace.h not found

UKMonkey, 2019-01-06:
With a fresh install of the arduino IDE; version 1.8.8 It seems that #include <pgmspace.h> should be replaced with

#ifdef AVR
#include <avr/pgmspace.h>
#else
#include <pgmspace.h>
#endif

Bodmer, 2019-01-06:
The TFT_eSPI library is not written to work with AVR processors, it will only work with the ESP8266 or ESP32 based boards.

If you do not select the ESP8266 or ESP32 as the board type you will get many error messages. Make sure that the ESP32 and/or ESP8266 board support packages are loaded and visibly to the IDE. Check this by looking at the list of boards available.

Post back if you still get error messages after the correct board has been selected. I am currently using 1.8.7

If you are writing generic sketches that can operate with other graphics libraries then you will have to include the AVR option.

UKMonkey, 2019-01-06:
Thanks - issue solved!

closed 2019-01-06 by UKMonkey