Bodmer/TFT_eSPI #282: RPI 3.5" LCD Screen Drawing issues (ESP32)

dcartman, 2019-01-07:
First, great library, thanks for putting this together.

I'm working on a clock, using RPI 3.5" LCD powered by an ESP32.

My sketch works, but after some time when the time digits are updated they are drawn about 100 pixels lower on the screen. Which leaves the lots of graphics tears on the screen. If I let it sit around longer, then the screen will stop drawing and just the white screen is presented.

Is anyone else running into issues where the screen turns white and stays white?

Bodmer, 2019-01-08:
It sounds like the data stream to the TFT is getting corrupted.

What SPI clock rate are you using?
Is there an example that comes with the library that exhibits the same problem?

Bodmer, 2019-01-22:
Check you power supply can handle the high current peaks during Wifi transmission.  If you have more info on your setup that I can test then I will re-open this issue.

closed 2019-01-22 by Bodmer