Bodmer/TFT_eSPI #283: ILI9341 Screen is only working with disabled Chip Select

tobimai, 2019-01-12:
Hi,

I have [this](https://www.banggood.com/2_8-Inch-ILI9341-240x320-SPI-TFT-LCD-Display-Touch-Panel-SPI-Serial-Port-Module-p-1206782.html?rmmds=myorder&cur_warehouse=CN) ILI9341 Display from Banggood and it works perfectly as long as I don't use ChipSelect by commenting out the`define TFT_CS 5` line. And connecting CS to GND. 

The problem is, that I want to use CS to be able to also use Touch. 
But when I use CS the Display just stays blank/white. 

I tried multiple Pins and Multiple Cables and multiple SPI Frequencies, nothing works. 
I also looked at the CS signal with an Oscilloscope and it's definitly doing something (seems like going low for a short time), I can't see more as it's an rather old analogue Oscilloscope. 

When I enable CS in the lib and connect CS to GND, the display also stays blank. 

I use an ESP32 DOIT v1 Devkit and Platformio for the programming. 

[Here](https://hastebin.com/kozibubapa.cpp) is the UserSetup.h and [Here](https://hastebin.com/tuyufopigi.cpp) is the main.cpp file wich is just the example slightly modified.

tobimai, 2019-01-15:
> Post a picture of your setup. Is the touch contoller chip wired to the ESP32 for these tests, if so was the touch controller chip select driven high so there is no SPI bus conflict. How are you powering the display?
> 
> These may sound like irrelevant questions but something is wrong with the setup, allocating a pin to the chip select but then not using it should not affect the behavior of the display.

I tested it on more pins and discovered, that CS only works on some Pins. 
Currently it works fine on 32 and 35. 

Could it have something to do with the settings of the internal pullUp-Resistors?

To your questions: 
When Testing, the Touch chip was disconnected and I am Powering the board from my Laptop (1.5A USB Port, shouldn't be a problem)

Bodmer, 2019-01-17:
GPIO 35 is input only so will not be able to drive the CS line. GPIO32 can be used and since it works I would keep using that pin.  I cannot replicate the issue with GPIO 5 using the Arduino environment, perhaps there is an issue with the library and the PlatformIO environment?

FLYINGSC0T, 2019-01-17:
Hi Bodmer. Thanks for a great library!! I have a problem getting Lora to work with an ILI9341 display on a TTGO T-Beam. If I comment out tft.begin() Lora works but if I enable it Lora does not work. The pin connections that I have set up in user setup .h file for the display are:

#define TFT_MISO 22  // Changed to 22 as GPIO 19 is used by Lora MISO on T-Beam
#define TFT_MOSI 25  // Changed to 25 as GPIO 23 is used ny Lora MOSI on T-Beam           
#define TFT_SCLK 21   // Changed to 21 as GPIO 18 is used by Lora CS on T-Beam
#define TFT_CS   13
#define TFT_DC    2
#define TFT_RST   4

Any idea where the confliction is ?
I have checked the GPIO's used by the T-Beam for the Lora module and I am sure they don't conflict but I can't find the problem.

BTW the display works perfectly using the above settings on an ESP32 Dev. board only that I can't get Lora and the display to work with the T-Beam.

Thanks in advance for any help.

George

achillhasler, 2019-01-18:
Hi George,
ESP32 have 2 usable SPI interface. (VSPI and HSPI) If you include <SPI.h> a variable SPI is defined whitch is used by the librarys.
If you dont create a new variable for the other SPI bus all librarys have to share the same bus (meens same connection for MOSI, MISO and SCK). Bud each dervice must have a own CS.
try:
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

ciao Achill

Bodmer, 2019-01-18:
Hi George, as Achill has said the Lora device and library use the VSPI interface in the ESP32, so if the TFT_eSPI libray moves the VSPI pins MOSI, MISO and SCK then the Lora function will stop working. The TFT CS and DC pins must be on unused pins.

I may add the option to select either HSPI or VSPI at a future date.

FLYINGSC0T, 2019-01-18:
Hi Achill,

Thanks for your reply.

The problem is that the T-Beam Lora uses the GPIO pins as follows:

MOSI     27
MISO     19
SCK          5
CS           18
RST         23

It looks like the library is moving the pins causing Lora to not work (?).

The GPIO pins I had set up for the display were:

 #define TFT_MISO 22   // Changed to 22 as GPIO 19 is used by Lora MISO on T-Beam
#define TFT_MOSI 25    // Changed to 25 as GPIO 23 is used by Lora MOSI on T-Beam
 #define TFT_SCLK  21    // Changed to 21 as GPIO 18 is used by Lora CS on T-Beam
#define TFT_CS      13
#define TFT_DC       2
#define TFT_RST      4

So, you mention that I could create a new variable for SPI to utilize the TFT_eSPI library.

How to I do that and which pins would you recommend to use specifically for the T-Beam ?

Maybe a stupid question, but can I pass as parameters SCK,  MISO, MOSI, CS in the call to tft.begin() because I only call tft.begin() with no parameters.

As the T-Beam is relatively new there is not a lot of information on the internet and I would really like to get this problem sorted.

Thanks again in advance for help you can offer.

Regards,

George

FLYINGSC0T, 2019-01-18:
Hi again Achill,

This is the code I’m using:

#include <SPI.h>
#include <LoRa.h>
#include <Wire.h>
#include "TFT_eSPI.h"

//  Define the LoRa module GPIO pins
#define LoRa_SCK      5     // GPIO5  -- SX1278's SCK
#define LoRa_MISO    19     // GPIO19 -- SX1278's MISO
#define LoRa_MOSI    27     // GPIO27 -- SX1278's MOSI
#define LoRa_NSS     18     // GPIO18 -- SX1278's CS
#define LoRa_RST     23     // GPIO14 -- SX1278's RESET
#define LoRa_DI0     26     // GPIO26 -- SX1278's IRQ(Interrupt Request)
#define Lora_BAND    868E6  // Frequency

unsigned int counter = 0;

// Define an instance of the TFT - uses hardware SPI
TFT_eSPI    tft = TFT_eSPI();

String rssi = "RSSI --";
String packSize = "--";
String packet ;

void setup()
{
  Serial.begin(115200);

  tft.begin();

  while (!Serial);
  Serial.println();
  Serial.println("LoRa Sender Test");

  SPI.begin(LoRa_SCK, LoRa_MISO, LoRa_MOSI, LoRa_NSS);

  LoRa.setPins(LoRa_NSS, LoRa_RST, LoRa_DI0);

  if (!LoRa.begin(Lora_BAND))
  {
    Serial.println("Starting LoRa failed !!");
    while (1);
  }
  Serial.println("Init OK !!");

  delay(1500);
}

void loop()
{
  Serial.print("Sending packet : ");
  Serial.println(String(counter));

  Serial.println(String(counter));

  // Send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(1000); // wait for a second
}

Regards,

George

Bodmer, 2019-01-18:
The key is to use the same pins for MISO, MOSI and SCK:

```
#define TFT_MISO 19 // Does not need to be conected to TFT unless reading from display
#define TFT_MOSI 27
#define TFT_SCLK  5
#define TFT_CS      13
#define TFT_DC       2
#define TFT_RST      4
```

FLYINGSC0T, 2019-01-18:
Hi Bodmer,

... on a normal ESP32 dev board that would be OK but the T-Beam doesn’t have pinout connections for 19, 27 and 5 !!

Thank you to reply.

Regards,

George

Bodmer, 2019-01-18:
Ok, in that case the only clean option is to add HSPI support which should be simple to do. I am extra busy at the moment but should be able to update the library within the next 10 days.

FLYINGSC0T, 2019-01-18:
Ok that’s great Bodmer !!

Not sure what HSPI is but I’ll look forward to an update.

Have a good weekend.

Regards,

George

Bodmer, 2019-01-19:
This site has good info and links for 
 the T-Beam board:
http://tinymicros.com/wiki/TTGO_T-Beam

FLYINGSC0T, 2019-01-19:
Thanks for the additional info Bodmer.

Regards,
George aka FlyingScot

http://tinymicros.com/wiki/TTGO_T-Beam

FLYINGSC0T, 2019-01-19:
> Hi George,
> have you tryed to share the spi-bus?
> [spi_share]<https://user-images.githubusercontent.com/46683972/51424547-a4e28680-1bcf-11e9-9c61-24dcf7238290.jpg>
> The lane near processor is a sd-card, away the tft.
> Both has connction to Ground, MISO, MOSI and SCK other are separate
> You see MOSI (D23) brown cable
> MISO (D19) violet cable
> SCK (D18) green cable
>
> on the jpg you see
> SD-Card use additional (D5) SS and 3.3V
> LCD use additional 5V (red cable), D4, RX2 (D16), TX2 (D17), RX0 (D3)
>
> The SD-Card needs no configuation because this are the default pins for VSPI.
>
> The User_setup.h for the shown picture is

Hi Achill,

Just to clarify, the board I’m using is a TTGO T-Beam and the pins you mention are not broken out on the board.

Thanks,

Have a good weekend also.

George

#define TFT_CS 3 // Chip select control pin
#define TFT_DC 4 // Data Command control pin
#define TFT_RST 17 // Reset pin
#define TOUCH_CS 16 // Chip select pin (T_CS) of touch screen (if you have a touchscreen)

nice Weekend

Bodmer, 2019-01-22:
@tobimai 

Your issue has got hijacked but I am still monitoring this thread if you wish to add more, or please start a new issue.

closed 2019-01-22 by Bodmer

FLYINGSC0T, 2019-01-22:
Hi,

Sorry but I don’t understand when you say my issue has got “hijacked”.

Please explain.

George

Bodmer, 2019-01-22:
Hi George,

I just meant that it got used to discuss other possible changes to the library unrelated to your problem!

FLYINGSC0T, 2019-01-22:
Hi Bodmer,

Ok no problem then.

I’ve another question - is it possible to connect a USB port to an ESP32 (T-Beam)?
I want to connect a USB software defined radio SDR to receive ADS-B transmissions from aircraft.

Thanks again.

Regards,
George

Bodmer, 2019-01-22:
The T-Beam beahves as a device.  I don't think you can use it as a host as ith would need software to drive your SDR.  You will probably find a Raspberry Pi will interface with the SDR.

FLYINGSC0T, 2019-01-22:
Ok. In fact I’ve already implemented SDR’s on a Raspi.

However, I have found that Olimex<https://www.olimex.com/Products/IoT/ESP32/ESP32-PRO/open-source-hardware> has produced an ESP32-Pro board with an ESP32 + USB port.

FLYINGSC0T, 2019-02-07:
Hi Bodmer,

Any progress on the implementation of HSPI for the T-Beam ?

Regards.

Ede2016, 2019-12-02:
Maybe this was answered already.
I use exactly the same display from Banggood with this code including CS pin and it works fine.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_master