Bodmer/TFT_eSPI #286: Add option to use HSPI interface

Bodmer, 2019-01-18:
The TFT_eSPI library uses the VSPI peripheral in the ESP32. There is a second HSPI peripheral available.

A user option couldbe added to alow a user to define which peripheral to use.  At the moment there is no clear justification to add this option when using the Arduino single tasking environment.

Bodmer, 2019-01-22:
Option added to use HSPI port instead of default VSPI port on ESP32. If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam) then add or uncomment the following line in the [setup file](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L268):

#define USE_HSPI_PORT

closed 2019-01-22 by Bodmer

cnc4less, 2019-06-30:
tried it with ili9341 also ili9486 still not working with the following pins

#define _cs   5   // goes to TFT CS
#define _dc   4  // goes to TFT DC
#define _mosi 23  // goes to TFT MOSI
#define _sclk 18  // goes to TFT SCK/CLK
#define _rst  32   // goes to TFT RESET

still works fine with adafruit lib please advise.

Bodmer, 2019-07-01:
Make sure you have edited the Setup.h file, or the selected user setup.

Get it running with the ILI9341 first.

Run the diagnostic "Read_User_Setup" example and view the serial output. It will list the settings and pin allocations the compiler is picking up from the setup file. Use a TFT_eSPI example for your TFT tests. The latest Adafruit examples may not run unmodified with TFT_eSPI.

cnc4less, 2019-07-14:
ran Read_User_Setup  got 
20:00:13.418 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:00:13.418 -> configsip: 0, SPIWP:0xee
20:00:13.418 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:00:13.418 -> mode:DIO, clock div:1
20:00:13.418 -> load:0x3fff0018,len:4
20:00:13.418 -> load:0x3fff001c,len:1100
20:00:13.418 -> load:0x40078000,len:9232
20:00:13.418 -> load:0x40080400,len:6400
20:00:13.418 -> entry 0x400806a8
20:00:14.282 -> ets Jun  8 2016 00:22:57