Bodmer/TFT_eSPI #287: Improve Touch implementation

achillhasler, 2019-01-18:
Hi Bodmer
i have improved the touch implementation.
- one configuration for all rotations
- stable x, y, z measure
- recognize if tuched with pen, finger or 2 fingers

changed files: Touch_calibrate.ino, Touch.h and Touch.cpp

[touch_new.zip](https://github.com/Bodmer/TFT_eSPI/files/2772660/touch_new.zip)

define TOUCH_IMPROVED in User_Setup.h and use the files from the zip.
Have Fun
Achill

Bodmer, 2019-01-18:
Hi Achill,
Thanks for this enhancement, the code looks very good and well written. I am not able to test at the moment.

I have been thinking about moving the touch handler code out of the TFT_eSPI library and create a new Arduino compatible library.  This is way down on my priority list so if you wish to facilitate this change, and host the library, then it would be great.  The library name I have in mind is TFT_eTouch, this is simply so the TFT_eSPI associated libraries are next to each other on the Arduino IDE library menu, but you can come up with your own name. In the longer term it would be great if the library can be expanded to support other touch controllers.

This approach has a number of benefits for me, in particular that it gives me time to focus on the graphics functionality and I have limited free time. This expansion library approach has been successfully taken for the TFTShape library.

I do not use touch screens on my Arduino projects for various reasons but prefer non-contact gesture sensors. A TFT_eSPI support library is on my "to do" list for this!

achillhasler, 2019-01-19:
Hello Bodmer,
there is a Touch library from Paul Stoffregen whitch works with ESP32
https://github.com/PaulStoffregen/XPT2046_Touchscreen
and one from Adafruit (https://github.com/adafruit/Adafruit_STMPE610) no ESP32 support

From the RaspberryPI there are some driver witch support touch
// linux sources
{linux_scr}/include/linux/spi/ad7877.h
{linux_scr}/include/linux/spi/ads7846.h

{linux_scr}/drivers/hid/hid-multitouch.c
{linux_scr}/drivers/drivers/input/touchscreen/ads7846.c (for all resistive chips)
-> support for TSC2046, ads7843, ads7845, AD7873 and AD7843

I think we can use the lib from Paul.
PS: Do you know the fbtft driver from RaspberryPI? ({linux_scr}/drivers/staging/fbtft) there a lot of tft definitions.

Bodmer, 2019-01-19:
Hi, yes I am aware of Paul's library and the Z pressure detect added in TFT_eSPI uses the same approach. I used that library briefly but Paul has little interest in supporting the Espressif products as he sells Teensies and I see a growing number of issues related to the ESP32 application which remain unanswered or unresolved. A fork could be created but the existing TFT_eSPI code appears to give more reliable touch validity, which is the main difficulty with the resistive touch screens.

Forking Paul's library and inserting your code is one way forward, but the API may have to change due to a different calibration approach, so then it becomes a new library.

I did look at the RPi TFT config files with a view to using them unmodified for different TFT drivers but gave up on that idea for various reasons, including concerns about change control and impacts.

mstrens, 2019-01-19:
Thanks for this new code.
I just got my display yesterday and I started some tests.
I tested the original code from Bodmer but I had issues because I regularly got false from getTouch() while the touchscreen was pressed. So, I could not detect in a proper way when the touch screen was really just released.
So, I tried this morning the new code.
Here a few remarks:
- I tried the provided ino file and changed the setupfile as requested for the new code. First I was unable to calibrate the display. The button in the corner becomes red but not green. Finally, I got it to work after  increasing from 8 up to 50 the value used to compare the min and max after 16 raw readings.
So, it could be that my display has much more noise than your.
- I made some tests changing the getTouchRaw() function in order to read more than twice (e.g. 8) each register and in order to take just the average of the last 2 conversions. I noticed that the accuracy of the positions becomes worst.
- I ask me if there is not an error in hasTouch()
There is : if (!getTouchRaw(xt, yt, lastTouchRawZ) && lastTouchRawZ > threshold) return false;
Should it not be : if ((!getTouchRaw(xt, yt, lastTouchRawZ) ) || lastTouchRawZ > threshold) return false; 
- In order to reduce the cpu time in the function getTouchRaw(), I will try to check the IRQ pin. I have access to this signal.

I do not really understand some code:
- what is the idea behind Z_LOCAL_MIN
- why filling lastTouchRawZ (and not z) with 0xFFFF in getTouchRaw()
- is the test (data1 == 0) in getTouchRaw() a good indicator that the touchscreen is not pressed? Otherwise, why having this check?

achillhasler, 2019-01-19:
Yes its getting red when not accurate. too much coffee? ;-)
-> we need a define for the 8 / 50 value

You have right, there i an error in my hasTouch() use
if (!getTouchRaw(xt, yt, lastTouchRawZ) || lastTouchRawZ > threshold) return false;
Thanks for reporting

when Z_LOCAL_MIN is NOT defined you get the first stabe z-Value
when defined we get some additional z-value til the new value is same or bigger. 
Example: fist z-Value : 1200, next 800, next 750, next 760 then you get 750 for z. When NOT defined 1200 for z. The idea comes form the Bodmer part. 
see: validTouch(uint16_t *x, uint16_t *y, uint16_t threshold)
  // Wait until pressure stops increasing to debounce pressure
  uint16_t z1 = 1;
  uint16_t z2 = 0;
  while (z1 > z2)
  {
    z2 = z1;
    z1 = getTouchRawZ();
    delay(1);
  }
if we define this the spi-bus is longer occupied. Your decision have a try.

lastTouchRawZ is a member var of the class whitch is accessible by getTouchRawZ(), z is not.
That means when (data1 == 0) after reading X (ctrl == 0b11010011) there is no touch and the return values x, y and z are not set and function returns false.
Bud getTouchRawZ() gives allways 0xffff when last call of hasTouch() or getTouchRaw() returns false.
PS: 
Coud be that you don't understand the break; in the do { } while() loop;
      if (data1 == 0) {
        lastTouchRawZ = 0xffff; // if 
        break; // has_touch stays false -> exit loop
      }
...
  } while (has_touch);

  SPI.transfer16(0x80); // set power down mode <-- after break this is executed and has_touch is still false 

by the way, the getTouchRaw() member should be private, call only hasTouch() for fetching values.
Also dont call hasTouch() more then 50 times each second. This is not handled by the lib.
You can add 3 additionals member Vars for x_screen & y_screen last_raw_mesure_time whitch you set in hasTouch() bevore return true.
At the beginning of hasTouch() check
if (millis() - last_raw_mesure_time < 1000/50) {
  if (lastTouchRawZ == 0xffff) return false;
  x = x_screen;
  y = y_screen;
  return true;
}
I dont have a Touch with the PENIRQ pin. i cant test this.

Can you implement that wir the interrupt and bring back the changed files?

@bodmer Your right the lib of Paul give only raw touch values and there is no configuration. We can do that better.
interface draft:
template <typename TFT>
class TFT_eTouch
{
public:
            TFT_eTouch(const TFT& display, uint8_t penirq = 0xff);

  bool      getXY(int16_t& x, int16_t& y, uint16_t threshold = 800);
  uint16_t getZ() const;

  bool      getCalibration(uint16_t *data, uint8_t xy_off) const;
  void      setCalibration(uint16_t *data);
 
  //  or  the data cold be a struct
  bool      getCalibration(TouchCallibation& data, uint8_t xy_off) const;
  void      setCalibration(const TouchCallibation& data);
};
rest will be private
using
TFT_eTouch<TFT_eSPI> touch(tft);
TFT_eTouch<Adafruit_ST7735> touch(tft); // for atmel
TFT_eTouch<ILI9341_t3> touch(tft); // for tennsy

tft need a Adafruit compatible iterface.

ciao Achill

Bodmer, 2019-01-19:
@Achill,
The template looks good. The touch chip select pin and option for penirq pin choice should be in the constructor, the chip select should be set high when the instance is created to avoid SPI bus conflicts during setup() init.

Ideally, the constructor should also accept a SPI interface choice, HSPI or VSPI for the ESP32, this could be alternatively be provided by a function call e.g getSPI() but then the library would be tied to TFT_eSPI which is not desirable if you wish use with other Adafruit GFX compatible graphics libraries. This ESP32 SPI bus selection feature is not compulsory but I am going to add a VSPI (default) or HSPI selection option to address issue #283 faced by @FLYINGSCOT on the TTGO T-Beam board.

Bodmer, 2019-01-20:
I think a good way to progress this is to create the new library and get it into a working state then refine it to work on the different setups. In the meanrime I wll leave the touch code in TFT_eSPI. Based on this strategy I am closing this issue since new ones can be created in the new repository.

I do have one minor change pending and that is multiple sampling in the raw function to allow for the ADC settle time.

A simple strategy I found works well for spurious detects that may occur on release of oresure is to not detect a touch for 20ms after the initial release, ie a debounce time. Hysterisis on the pressure detect could be another option.

closed 2019-01-20 by Bodmer