Bodmer/TFT_eSPI #288: building

bigcattus, 2019-01-19:
I am having a lot of problems building the library (using a makefile).  Looking for example at
/mnt/data/electronics/microcontrollers/esp8266/Other/TFT_eSPI/Extensions/Sprite.cpp:

Smooth_font.cpp:13:6: error: 'TFT_eSPI' has not been declared
 void TFT_eSPI::loadFont(String fontName)

Looking at Smooth_font.cpp, there is no include file that would bring this symbol in.

Also, spaces in the directory path names (e.g. in examples folder) make things more difficult to handle.

Thanks.

Bodmer, 2019-01-19:
The library has been developed to be compiled under the Arduino IDE. I am not able to support other IDE environments, if you find a solution then do generate a pull request and provide a help file. I will remove the spaces in the example folder names as I can see that would be an issue.

closed 2019-01-19 by Bodmer

bigcattus, 2019-01-20:
Understood.  I can certainly refactor things so that things build, but
I am not sure you would like the changes.  The biggest problem is the
inclusion of .cpp, .c files either into header files, or into other
.cpp files.   For example, Sprite.cpp does not see the class header
and does not build.  On the other hand, attempting to build the source
where .cpp files are included into other files as a library will
result in a linking problem (multiple symbol definitions error).
Sprite.cpp would either have to be combined with Sprite.h, or
renamed to something like Sprite.inc so that a generic Makefile would
leave it alone.   It is possible to write a custom project Makefile
that would resolve some of the issues, but that would be a bad idea
because most people's make-based build systems rely on the above
conventions.

Please don't take this as a criticism of your efforts.  I certainly
understand the scope of your work (Arduino IDE) and that things run
fine in that case.  Just wanted to point out the problems I ran into.

Thank you.

Bodmer, 2019-01-20:
OK, I can understand the difficulty and you are right I would not like the changes! The library should compile under the PlatformIO environment as that handles the Arduino environment coding.

Thanks for your insight.