Bodmer/TFT_eSPI #290: Processing to convert Glyph font (line-awesome)

ismaelit, 2019-01-19:
I have successful converted .ttf and .otf fonts to .vlw before. But now I am trying to convert the glyphs of Line-Awesome (https://icons8.com/line-awesome), and ever trying to convert the unicode blocks from 0x0000 to 0xFFFF, I can't see any character, only boxes. 

Is there an issue or the data blocks of the glyphs are in other strange block range?

Regards.

Bodmer, 2019-01-20:
It info provided indicates the glyphs are in Unicode range 0xf101 to 0xf38d but there are gaps, any Unicode that does not have a glyph in the set will show as a box by convention.  Have you tried just that range?
I am not able to run tests myself at the moment.
The line-awesome icons look useful, thanks for the link.

ismaelit, 2019-01-20:
There is something in this range! But It looks incomplete and have a lot of duplicates. How can I look for other ranges? 

[image]

Bodmer, 2019-01-20:
Check the font documentation here:
https://icons8.com/line-awesome/cheatsheet

It looks like there are duplicates in the font and you have the complete set but I have not looked that carefully.

closed 2019-01-20 by Bodmer