Bodmer/TFT_eSPI #291: Request SSD2119 support

Blom1974, 2019-01-20:
Would be nice to have SPI - SSD2119 support. Attached a sample init code. And link to the controller data sheet.
https://www.crystalfontz.com/controllers/SolomonSystech/SSD2119/352/

`//INITIALIZATION 
void main_init(void)
{   RES=1;
	delayms(1);	
	RES=0;
	delayms(5);
	RES=1;
	delayms(5);
 
	main_W_com_data(0x0028,0x0006);    // VCOM OTP
       // Page 55-56 of SSD2119 datasheet

		main_W_com_data(0x0000,0x0001);    // start Oscillator
       // Page 36 of SSD2119 datasheet
	
		main_W_com_data(0x0010,0x0000);    // Sleep mode
       // Page 49 of SSD2119 datasheet
	
		main_W_com_data(0x0001,0x72EF);    // Driver Output Control
       // Page 36-39 of SSD2119 datasheet
	
		main_W_com_data(0x0002,0x0600);    // LCD Driving Waveform Control
       // Page 40-42 of SSD2119 datasheet
	
		main_W_com_data(0x0003,0x6A38);    // Power Control 1
       // Page 43-44 of SSD2119 datasheet
	
	    main_W_com_data(0x0011,0x6270);    // Entry Mode   Horizontal screen mode   320*240    65k colour
	   // Write_Cmd_Data(0x0011,0x6218);  	//   Vertical screen mode     240*320   65k colour
       // Page 50-52 of SSD2119 datasheet
	
		main_W_com_data(0X000F,0x0000);    // Gate Scan Position
       // Page 49 of SSD2119 datasheet
	
		main_W_com_data(0X000B,0x5308);    // Frame Cycle Control
       // Page 45 of SSD2119 datasheet
	
		main_W_com_data(0x000C,0x0003);    // Power Control 2
       // Page 47 of SSD2119 datasheet
	
		main_W_com_data(0x000D,0x000A);    // Power Control 3
       // Page 48 of SSD2119 datasheet
	
		main_W_com_data(0x000E,0x2E00);    // Power Control 4
       // Page 48 of SSD2119 datasheet
	
		main_W_com_data(0x001E,0x00be);    // Power Control 5
       // Page 53 of SSD2119 datasheet
	
		main_W_com_data(0x0025,0x8000);    // Frame Frequency Control
       // Page 53 of SSD2119 datasheet
	
		main_W_com_data(0x0026,0x7800);    // Analog setting
       // Page 54 of SSD2119 datasheet
	
		main_W_com_data(0x004E,0x0000);    // Ram Address Set
       // Page 58 of SSD2119 datasheet
	
		main_W_com_data(0x004F,0x0000);    // Ram Address Set
       // Page 58 of SSD2119 datasheet
	
		main_W_com_data(0x0012,0x08D9);    // Sleep mode
       // Page 49 of SSD2119 datasheet
	

		// Gamma Control (R30h to R3Bh) -- Page 56 of SSD2119 datasheet
		main_W_com_data(0x0030,0x0000);
	
		main_W_com_data(0x0031,0x0104);
	
		main_W_com_data(0x0032,0x0100);
	
		main_W_com_data(0x0033,0x0305);
	
		main_W_com_data(0x0034,0x0505);
	
		main_W_com_data(0x0035,0x0305);
	
		main_W_com_data(0x0036,0x0707);
	
		main_W_com_data(0x0037,0x0300);
	
		main_W_com_data(0x003A,0x1200);
	
		main_W_com_data(0x003B,0x0800);
		 	
		main_W_com_data(0x0007,0x0033);    // Display Control 
      // Page 45 of SSD2119 datasheet
	
		 delayms(100);
	
		main_Write_COM(0x0022);    // RAM data write/read
}`

Blom1974, 2019-01-21:
> I do not have one of those displays so that makes debugging an issue. Can you privide a web link to an SPI display that is on a breakout pcb?
> 
> I prefer to only support commonly available cheap displays rather than one-offs due to the time needed to add/debug new display controllers.
> 
> If you wish to fork the library and add the display driver yourself then I can provide instructions on what needs to be added and where. The controller code can be added to the master library via a pull request.

Thanks for the answer. Haven´t seen these on a breakout board. Understand that it is hard without a display. I got a couple of different displays from Powertip with this controller so I am able to test this myself. Will be happy to Fork and test it out myself if you are able to provide some instructions to get me going. Thanks in advance.

Bodmer, 2019-01-21:
Hi,,
Create 3 header files [here](https://github.com/Bodmer/TFT_eSPI/tree/master/TFT_Drivers). Named:

SSD2119_Defines.h  - linked in below [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L109-L111).
SSD2119_Init.h - linked in below [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L378-L379).
SSD2119_Rotation.h - linked in below [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L437-L438).

Add driver option to setup [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L30) and [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/SetupX_Template.h#L30).

Use for example the ILI9341 files as an example.

Bodmer, 2019-01-22:
I will close this for now.

You can continue to ask questions on this issue thread.

closed 2019-01-22 by Bodmer