Bodmer/TFT_eSPI #292: Enhanced version of touchscreen

mstrens, 2019-01-21:


mstrens, 2019-01-21:
I use this model
https://fr.aliexpress.com/item/3-2-pouce-320-240-S-rie-SPI-TFT-LCD-Module-D-affichage-avec-cran-Tactile/32960934541.html?spm=a2g0s.9042311.0.0.799b6c37iCxsnC

I think it has a XPT2046 chip which is the same (as far I know) as a Texas Instrument chip.
I do not know why the IRQ signal is wrong.

mstrens, 2019-01-22:
FYI, I removed the first "for" loop (4 times dummy reading of the chip) and it still works well.
So the delays to execute the getTouch is now divided by 2 (about 50 us when not pressed and 120 when pressed).

Bodmer, 2019-01-22:
The zip file version above is not a direct replacement, it generates errors on compile. A new supporting header Touch.h is needed. Make sure the existing examples continue to run.

I have have marked as a development discussion and will close it to avoid confusion.

Do continue the discussions.

Bodmer, 2019-01-31:
I updated the library a while ago to multi sample  x sand y. It does seem to improve the raw touch read consistency but it apears to have not effect after filtering by getTouch(). YMMV.

mstrens, 2019-01-31:
If I understand correctly the application note, when you perform multiple conversions, you should use commands with the last significant bit set on 1 (in order to keep power on and so getting more stable voltages to read).
So e.g. instead of sending a command "0xDO", you should send "0xD1" (except for the last command sent in the function).
In order to reduce the noise on the signal, the code I wrote does not only performs multiple reads  (I make 4 successives reads for X and for Y) but it also calculates the min and max and check that the difference between min and max is less than a parameter. If not ok, it performs a new set of 4 reads. if ok, it calculates the average (using just sum >> 2 which is the same as sum/4).

Bodmer, 2019-02-02:
I use the[ XPT2046 data sheet here](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwiosYLIv53gAhW6VRUIHQOVBv4QFjAAegQIChAC&url=https%3A%2F%2Fwww.buydisplay.com%2Fdownload%2Fic%2FXPT2046.pdf&usg=AOvVaw2CpjSQGkSlrKYT-_WC0WNe). Page 24 and Table 8 indicate the ADC is available instantly and thus there is not need to control PD0.

achillhasler, 2019-02-03:
I have done my work see https://github.com/achillhasler/TFT_eTouch for new tochscreen dirver. In the User_Setup.h of TFT_eSPI you have to disable
//#define TOUCH_CS 16     // Chip select pin (T_CS) of touch screen

This CS is now used by the constructor
TFT_eTouch<TFT_eSPI> touch(tft, 16);

If you need documentation install doxygen and graphviz and build the documentation with the Doxyfile in directory docs

have a try
ciao Achill

Bodmer, 2019-02-03:
Thanks. looks good.

I will create a test setup, but it will be a while before I can test it.

Bodmer, 2019-02-11:
I have tried the library with an ESP8266 and ILI9341 screen with an XPT2046 fitted.

The raw sketch runs fine and seems to generate sensible values.

At the moment the ESP8266 just keep rebooting on the watchdog when trying the calibration sketch. 

I know there is a bug in the TFT_eSPI library which is stopping the built in touch controller from working so that is not helping. At the moment the TFT_eSPI library indicates the screen is being touched all the time.

 I will sort the TFT_eSPI issue first and then come back to test the TFT_eTouch library again.

Bodmer, 2019-02-14:
I have repaired TFT_eSPI (it was stuck in SPI write only mode on the ESP8266).

I have only tried the TFT_eTouch library with an ESP and ILI9341 screen which is quite noisy for output position.  The raw function works fine and gives stable outputs.

The calibrate sketch puts  marker on the screen then the ESP8266 reboots on the watchdog.  I do not have the Penirq wired.

Have you tested the library on an ESP8266?

achillhasler, 2019-03-03:
I have updated TFT_eTouch to ver 00.05.00 but now i need a interface change in TFT_eSPI.
Add static getSPIinstance() function to TFT_eSPI class, for getting spi bus used by TFT_eTouch.
On SPI default config, (VSPI Port) do not create a new var, use the SPI var exported by SPI.h Header file. 
TFT_eSPI.h
`  void     getSetup(setup_t& tft_settings); // Sketch provides the instance to populate
+#ifndef ESP32_PARALLEL
+  static SPIClass& getSPIinstance();
+#endif
  int32_t  cursor_x, cursor_y, padX;
`
TFT_eSPI.cpp
`#if defined (ESP32)
  #if !defined (ESP32_PARALLEL)
    #ifdef USE_HSPI_PORT
      SPIClass spi(HSPI);
    #else // use default VSPI port
+-      SPIClass& spi = SPI; // use the predefined SPI variable which is defined by SPI.h
    #endif
  #endif
#else // ESP8266
+-  SPIClass& spi = SPI; // use the predefined SPI variable which is defined by SPI.h
#endif
+#ifndef ESP32_PARALLEL
+SPIClass& TFT_eSPI::getSPIinstance() { return spi; }
+#endif
`
The Conways_Life example supports yours and my touch implementation to see the difference.
With the edit_calibration example we can adjust the configuration and store it in FLASH. only tested on 480*320 display.

Bodmer, 2019-03-03:
I have mode the changes to my copy.

The calibrate sketch throws compile errors due to some instance naming errors. When corrected the ESP8266 keeps rebooting.

Conways_Life example compiles but the ESP8266 keeps rebooting immediately after the calibration screen is drawn.

raw example compiles but never reports anything to the serial port, again I am using an ESP8266 and ILI9341 screeen.

achillhasler, 2019-03-04:
I test that on a ESP32. i dont have a ESP8266, but why does it rebooting. Difficult to test on ESP32 there is no reboot. I test with a RPI_ILI9486_DRIVER display.

If you define TOUCH_CS in TFT_eSPI User_Setup.h my touch implementation is not used in the Connways_Life demo. Does ist also reboot on ESP8266?

Bodmer, 2019-03-05:
OK, I will create a setup with an ESP32 and touch screen in the next few days and see how it goes.

achillhasler, 2019-06-09:
There is a new version of TFT_eTouch whitch support ESP8266 chips. Have a try

xfdr0805, 2020-06-19:
> There is a new version of TFT_eTouch whitch support ESP8266 chips. Have a try

https://github.com/achillhasler/TFT_eTouch/issues/5

need your help ,tks!

closed 2020-12-16 by Bodmer