Bodmer/TFT_eSPI #293: Board Manager esp8266 version above 2.4.1 & issues with unicode/utf-8

ismaelit, 2019-01-21:
When I use a boards manager version of the Arduino IDE for esp8266 higher than 2.4.1, I have problems with the use of unicode / utf-8 characters, they simply are not printed on the screen, and no characters are printed after the special character.

Does the TFT_eSPI library suggest the use of version 2.4.1 even, or can this be a local problem of mine? I tried on two different computers, and so it did.

Using board Wemos D1 Mini. 

[image]

Bodmer, 2019-01-22:
I am using 2.4.2 and it is working OK.

There is a bug in some versions of the Arduino IDE which fails to compile the correct Unicode values unless the sketch is saved first. One way to ensure this happens automatically is to tick the "Preferences" box "Save when verifying and uploading".

Run the Unicode_Test sketch, make sure the fonts have been uploaded to SPIFFS, with this example you can check the Serial Monitor output reports font files are found.

If your own sketch does not work post the entire sketch and font file so I can test and debug for you.

Bodmer, 2019-01-22:
I have just run the sketch and I see lines 135 and 138 do not print to screen.

I will investigate.

Thanks for reporting this.

Bodmer, 2019-01-22:
You are right, there is a bug in the ESP8266 board package 2.4.2 and later which fails to build Unicode values into a string.

I cannot fix this in the library so please use **ESP8266 board core version 2.4.1** until the core developers have fixed the issue.  I will report it.

Thanks for pointing out this issue.

Bodmer, 2019-01-22:
OK, I have tracked this down to the TFT_eSPI library, the way the print stream is handled has changed.

It is now fixed. Thanks for reporting this.

commit b5db54dc (2019-01-22): Fix #293 for ESP8266 core 2.4.2 and later

UTFT8 encoded Unicode values were not being drawn on screen with versions after 2.4.1 of the ESP8266 core.
commit 8cf85c89 (2019-01-22): Fix #293 for Sprites
closed 2019-01-22 by Bodmer