Bodmer/TFT_eSPI #294: ESP32 HSPI pins + ILI9341

lunaoe, 2019-01-23:
Hi!

I'm using a ESP-WROOM-32 with ILI9341, nothing else connected, I want to use HSPI. Looking at ESP32 pinouts I modified User_Setup.h like this:

`#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    26  // Data Command control pin
#define TFT_RST   27  // Reset pin (could connect to RST pin)

#define USE_HSPI_PORT
#define SUPPORT_TRANSACTIONS`

But it doesn't work. Display stays white. What's wrong? Thanks

Bodmer, 2019-01-23:
Have you got the display working at all in the past?

Try with #define USE_HSPI_PORT commented out.

I have tried a setup with an ESP32 board using the exact same pins and it works with the default VSPI and also the HSPI mode. The SPI port only matters if there is something using the other port.

Run the "Read_User_Setup" diagnostic to make sure the compiler is picking up the right pins. My output is:

```
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D13 (GPIO 13)
MISO    = D12 (GPIO 12)
SCK     = D14 (GPIO 14)
TFT_CS   = D15 (GPIO 15)
TFT_DC   = D32 (GPIO 32)
TFT_RST  = D27 (GPIO 27)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.0 MHz 
```
Post a picture of your setup and a link to the vendors web page for the display. Are you sure it is a ILI9341?

Bodmer, 2019-01-23:
I had DC on pin 32 but moved it to 26 and it still runs with the new configuration:
```
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D13 (GPIO 13)
MISO    = D12 (GPIO 12)
SCK     = D14 (GPIO 14)
TFT_CS   = D15 (GPIO 15)
TFT_DC   = D26 (GPIO 26)
TFT_RST  = D27 (GPIO 27)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.0 MHz 
```

lunaoe, 2019-01-23:
It works if I comment the TFT_CS line and connect CS to GND.
`//#define TFT_CS   15`

lunaoe, 2019-01-23:
This is with TFT_CS line commented (works fine)
```
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D13 (GPIO 13)
MISO    = D12 (GPIO 12)
SCK     = D14 (GPIO 14)
TFT_DC   = D26 (GPIO 26)
TFT_RST  = D27 (GPIO 27)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.0 MHz 
```

This is with TFT_CS set to 15. It doesn't work, I tried connecting CS to GPIO15 and GND, same result on both, white screen.
```
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D13 (GPIO 13)
MISO    = D12 (GPIO 12)
SCK     = D14 (GPIO 14)
TFT_CS   = D15 (GPIO 15)
TFT_DC   = D26 (GPIO 26)
TFT_RST  = D27 (GPIO 27)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.0 MHz 
```

Bodmer, 2019-01-23:
I do not know why that would happen. Have you tried any other pins for the TFT_CS line? You could use pin 25 as that is much nearer the rest of the pins you are using. Does that work?

Which board have you selected in the IDE?

lunaoe, 2019-01-23:
This is the config I use in Arduino IDE. Using 'esp32 by Espressif Systems version 1.0.1'

[image]

I'll try using a different pin for CS, I'll post the result later. I think this problem also happens with VSPI because it was the first thing I tried and never got the display to work. I'm checking that again.

Bodmer, 2019-01-23:
You have used the same board I selected. I have upgraded to 1.0.1 of the ESP board package but I see no difference, it is still working.

Thanks for doing all these tests, it is as if the TFT_CS line is not being driven. I have just connected an oscilloscope to mine and the CS line is toggling.

The only other thing I can think of is that the display is not being initialized for some reason.

I have tried another 2.8" ILI9341 display and that one works too.

**I have slowed down the CS line and uploaded a new version for you to try. Only the TFT_eSPI .cpp and TFT_eSPI.h files have changed.**

lunaoe, 2019-01-24:
I added a LED to CS and I can see it is toggling. (TFT_CS set to pin 25).
[image]
But still white screen

I updated TFT_eSPI.cpp and TFT_eSPI.h and reuploaded code:
[image]
It now works. 
I don't have an oscilloscope, I wonder how it looks now vs before on my setup.

Bodmer, 2019-01-24:
Great to hear it is now working.

I have examined the code in more detail since the last upload and have made some changes to the strobe order for the CS and DC lines because DC naturally gets a long setup time as it only need to be valid for the last bit.  It appears that the setup time from CS falling low needs to be longer on your display.

I will be uploading the new image within the next 24 hrs and it will be helpful if you an test it for me.

lunaoe, 2019-01-25:
New code works as well.

Old code with TFT_CS disabled and connected to GND
[image]

First code that worked 
[image]

New code
[image]

Bodmer, 2019-01-25:
Thanks for testing, I am pleased it is working.

Some time results are higher numbers because the library now draws extra rectangles that were mised out previiusly because they started with off screen coordinates. The library now crops the rectangular area and draws the portion the would be on screen, so now more pixels are being drawn.

closed 2019-01-25 by Bodmer

Bodmer, 2019-01-25:
Looks like pixels are missing and there is a distinct pattern sugesting there is a periodic electrical disturbance.

I will see if I can reproduce the problem, it will be a few days before I am able to do this.

I see from earlier pictures that you have added decoupling caps, I have not needed them. Is there a reason why you added them? Does changing the way the setup is powered changes the behaviour?

Bodmer, 2019-01-26:
Can you try the new PDQ graphics test as that holds CS low during all pixel writes to screen.

Which version of the IDE board package are you using, I think the latest is 1.0.1 but that seems to have a lot of new issues including glitches on the reset line.

Bodmer, 2019-01-31:
I have run some tests on my display and even at 80MHz SPI rate the PDQ pixels test is perfect so I am not sure what I can do to help. If you manage to identify the cause then do post back.

gojimmypi, 2019-06-09:
@Bodmer and @lunaoe - thanks for posting these details and results.

In using [these pins](https://github.com/Bodmer/TFT_eSPI/issues/294#issuecomment-456948513) (in both steps 1 and 5), this is _software_ SPI (SWSPI), no? 

```
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC   26  // Data Command control pin
#define TFT_RST  27  // Reset pin (could connect to RST pin)
```
Isn't the fast, hardware SPI (HWSPI) the topic of this issue 'HSPI'  (`HSPI` [by definition](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino#L37) uses MOSI=13, no?) at noted:

```
//SCLK = 14, MISO = 12, MOSI = 13, SS = 15
```

I ask, as I have a [similar problem](https://github.com/emard/ulx3s/issues/8) in getting the SPI driver to work properly with the non-default `HSPI`.

I thought it would be simply a matter of editing [this last line in the SPI.cpp file](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.cpp#L295):

```
SPIClass SPI(HSPI);
```
but no luck, as I have speckled, multi-color snow on the display and the code is frozen at `display.fillScreen(LED_WHITE_HIGH)` (using the Adafruit drivers)