Bodmer/TFT_eSPI #296: esp8266 ili9341 touch creates artefacts

mocholor, 2019-01-30:
after adding touch functionality (i.e. invoking tft.getTouch) I get strange artefacts (see red arrows) on the display. (touch works...)
When I remove the tft.getTouch everything looks fine again.

[image]

Bodmer, 2019-01-31:
Run the diagnostic sketch Read_User_Setup and post the results here so I can see the configuration you are using. The result you are getting imply that the TFT chip sekect is stuck low and tge TFT is receiving SPI data destined for tge touch controller.

mocholor, 2019-01-31:
hmmm. Actually my TFT_CS is tied to GND. Only TOUCH_CS is wired. So as you said, TFT receives some stuff it shouldn't.
Read_User_Setup throws strange errors due to hardwareserial::printf
So here's the interesting part of User_Setup.h. (where my interpreation was you COULD tie TFT_CS to ground)

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP8266 SETUP ######

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

//#define TFT_BL PIN_D1  // LED back-light (only for ST7789 with backlight control pin)

//#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen

//#define TFT_WR PIN_D2       // Write strobe for modified Raspberry Pi TFT only

// ######  FOR ESP8266 OVERLAP MODE EDIT THE PIN NUMBERS IN THE FOLLOWING LINES  ######

// Overlap mode shares the ESP8266 FLASH SPI bus with the TFT so has a performance impact
// but saves pins for other functions.
// Use NodeMCU SD0=MISO, SD1=MOSI, CLK=SCLK to connect to TFT in overlap mode

// In ESP8266 overlap mode the following must be defined
//#define TFT_SPI_OVERLAP

// In ESP8266 overlap mode the TFT chip select MUST connect to pin D3
//#define TFT_CS   PIN_D3
//#define TFT_DC   PIN_D5  // Data Command control pin
//#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1  // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

Bodmer, 2019-01-31:
A pin must be allocated and connected to the TFT chip select line and another different pin to the touch controller chip select line. It should then work fine.

Bodmer, 2019-01-31:
This is the arrangement needed (in this case there are 3 slaves):

https://upload.wikimedia.org/wikipedia/commons/f/fc/SPI_three_slaves.svg

mocholor, 2019-01-31:
since I'm running out of pins: would connecting CS and T_CS via an inverter to - what GPIO - work??

Bodmer, 2019-01-31:
Choose an ESP8266 pin for the touch CS and connect to the touch controller CS, then the inverted signal to the TFT CS.

closed 2019-01-31 by Bodmer