Bodmer/TFT_eSPI #298: Display flashes on M5Stack-fire on startup

an-erd, 2019-02-04:
I updated the M5Stack library with your current code. This was needed because an older version of the code is used in the M5STack library, and there'd been a change in TFT_eSPI library to use PSRAM. Everything is working fine now, but there's ony issue I can't resolve.

During startup/initialization the display flashes for a short time (~200ms) from black/off to a white screen and back to black again. Any idea on how to remove this flashing or where to start looking at?

an-erd, 2019-02-04:
I think that it is to early to turn on the backlight during TFT_eSPI::init() (https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L350), because there's no change to clear the background to TFT_BLACK. Is there a change to initialize the display to black before calling BL on?

Bodmer, 2019-02-04:
One way to stop this happening is to not define the backlight control pin by [commenting out the line here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup12_M5Stack.h#L15).

You must then control the pin from your sketch.

At the next update I will move the backlight switch on to the last operation during the init period.

closed 2019-02-04 by Bodmer

Bodmer, 2019-02-08:
I updated the library and have moved the LED back light switch on to the last operation during init().

Thanks for reporting this.