Bodmer/TFT_eSPI #299: ILI9486 Screen with ESP32 SPI and Parallel

HumamHelfawi, 2019-02-04:
Hi,
I have this screen (Driver: ILI9486):
[image]

I tried to connect it in parallel as it was decided here https://github.com/Bodmer/TFT_eSPI/issues/167, I did not get any results.

I connect it using SPI and as it was decibed in Setup11_RPi_touch_ILI9486. I did not get any results.

Is there any other suggestion?

Thank you very much!

Bodmer, 2019-02-04:
Have a look at [this thread](https://forum.arduino.cc/index.php?topic=540147.0).

Bodmer, 2019-02-04:
If you wish to try the pseudo 8 bit mode with the TFT_eSPI library then use:

  #define PSEUDO_8_BIT

in the setup file. I have neot tried it myself, so cannot guarantee it will work.

HumamHelfawi, 2019-02-05:
Yes the colors are much more realistic now. Thanks again Bodmer!

flykarlos, 2021-03-24:
dear HumamHelfawi and Bodmer
I have few questions regarding connecting display mentioned in first post.
1. 

> Thank you very much for your reply!
> 
> Just in case someone passes by this issue in the future. I managed to to run it using this settings:
> 
> ```
> #define ILI9488_DRIVER
> #define ESP32_PARALLEL
> #define TFT_CS   15  // Chip select control pin (library pulls permanently low
> #define TFT_DC   2  // Data Command control pin - use a pin in the range 0-31
> #define TFT_RST  -1  // Reset pin, toggles on startup
> #define TFT_WR   4  // Write strobe control pin - use a pin in the range 0-31
> #define TFT_RD    3  // Read strobe control pin  - use a pin in the range 0-31
> #define TFT_D0   22  // Must use pins in the range 0-31 for the data bus
> #define TFT_D1   13  // so a single register write sets/clears all bits.
> #define TFT_D2    9  // Pins can be randomly assigned, this does not affect
> #define TFT_D3   0  // TFT screen update performance.
> #define TFT_D4   17
> #define TFT_D5   16
> #define TFT_D6   27
> #define TFT_D7   14
> ```
> 
> The screen is almost blue which is logical since only the first 8 bits are connected.

Working config from this post has DC pin defined, but display does not have one. So which pin on display is DC?

2. Should I connect also MOSI MISO and SCLK for SPI initialization?
thanks!

Bodmer, 2021-03-24:
No, those SPI connections only connect to the SD card slot.

The RS pin has the same function as the DC pin, it is just named differently.

flykarlos, 2021-03-25:
Thanks Bodmer, as usual You are extremely helpful. Can You advice about RD pin? Such pin marking is not present on this display.. 
And should I also `#define TFT_PARALLEL_8_BIT` before ` #define ESP32_PARALLEL`? I tried many different  pin mappings and settings and it does not work. I have exactly the same display as in first post. I am using  ESP32-WROVER-B. Thanks in advance!

drewp, 2021-03-25:
DC is sometimes called RS. My board is like http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO and calls it LCD_RS. 

The datasheet at https://www.waveshare.com/w/upload/7/78/ILI9486_Datasheet.pdf calls it "D/CX".

I'm working on an esphome driver for these 8-bit boards, and I'm using DC internally since I can remember "data/command mode".

drewp, 2021-03-25:
Sorry, didn't refresh this page today before answering. RD is "RDX" in the spec, and the diagram on page 9 shows "WRX/SCL" as one input. SCL is the spi clock, so I'd guess your board's "CLK" is where to send the RD signal.

Bodmer, 2021-03-26:
It looks like the display is write only, so the RD pin can be left unconnected.

drewp, 2021-03-26:
Mine had to be held at RD=high for communication to work. HTH

I'm up and running in esphome now!
[image]

flykarlos, 2021-03-26:
Thanks guys!
 I managed to make it work with pseudo 8bit mode. But I have a problem displaying text:
[image]
Image shows below program drawing text "TEST" on sprite pushed to tft. Text looks distorted, almost like angled by some strange angle.

```
#include <TFT_eSPI.h>                
TFT_eSPI    tft = TFT_eSPI();         
TFT_eSprite spr = TFT_eSprite(&tft);

int colors_data[12]={0x0000, 0x1818, 0xE0E0, 0x0707, 0x7777, 0xF8F8, 0xE7E7, 0xFFFF, 0xB5B5, 0x6B6B, 0xF4F4, 0xE4E4};
int i = 0;

void setup()
{
  tft.init();
  spr.setColorDepth(8);
  spr.createSprite(200, 200);
    spr.setTextColor(TFT_YELLOW, TFT_BLACK);
}

void loop(void)
{
  tft.fillScreen(colors_data[i]);
  spr.drawString("TEST",100 ,100, 2);
  spr.pushSprite(60, 140);
  i = (i == 12) ? 0 : i+1;
  delay(2000);
}

```

 Thanks for help!

flykarlos, 2021-03-26:
Ok, disregard last post. Seems that `spr.setColorDepth(8);` was causing the problem. Removed it and now image looks better..

Bodmer, 2021-03-26:
You should be able to set the colour depth to 1, 4, 8 and 16 bpp, this works OK on my SPI display.

I do not have a psuedo16 bit setup to test so maybe it is just that interface that is the problem?

flykarlos, 2021-03-27:
Thanks Bodmer for Your efforts to help me each time I am writing here with a problem. 
When I use color depth settings other than 16 bpp the sprite used with this setting goes black.
I reinstalled library to make sure my problems are not caused by my modifications done to library.
I am using this user_setup.h:

```
#define ILI9486_DRIVER
#define ESP32_PARALLEL
#define PSEUDO_16_BIT

#define TFT_CS   19  
#define TFT_DC  3 
#define TFT_RST  15 
#define TFT_WR   18  
#define TFT_RD    -1  
#define TFT_D0   4
#define TFT_D1   26 
#define TFT_D2    2
#define TFT_D3   14 
#define TFT_D4   13
#define TFT_D5   12
#define TFT_D6   25
#define TFT_D7   27

#define LOAD_GLCD   
#define LOAD_FONT2 
#define LOAD_FONT4  
#define LOAD_FONT6 
#define LOAD_FONT7
#define LOAD_FONT8  
#define LOAD_FONT8N
#define LOAD_GFXFF

#define SMOOTH_FONT
```

I am using Rotated_Sprite_1.ino  example with addition of color definitions supporting PSEUDO_16_BIT mode:

```
#define TFT_BLACK   0x0000
#define TFT_BLUE    0x1818
#define TFT_RED     0xE0E0
#define TFT_GREEN   0x0707
#define TFT_CYAN    0x7777
#define TFT_MAGENTA 0xF8F8
#define TFT_YELLOW  0xE7E7
#define TFT_WHITE   0xFFFF
#define TFT_LIGHT_GREY  0xB5B5
#define TFT_DARKGREY   0x6B6B
#define TFT_PINK    0xF4F4
#define TFT_ORANGE  0xE4E4
```
I wired display D8-D15 pins as you described [here](https://forum.arduino.cc/index.php?topic=540147.0)
If I use this or any other example with text there is distortion in displayed image. It looks like each second pixel line is missing:

[image]
[image]

on the other hand same example displays numbers correctly(note: blue line is most probably caused by wiring interference):
[image]

thanks for advices!

Bodmer, 2021-03-27:
OK, I just realised that the pseudo 16 bit mode must have been removed at some point. Not sure which version I did this but some earlier versions of TFT_eSPI did have this feature...

The problem is that every pixel (except single drawPixel where it does not matter) is being written twice. As the library expects to write 2 x 8 bit values it generates 2 write stobes per pixel. So, what I need to do is reinstate that mode. I will look to see if that is easy to do.

flykarlos, 2021-03-27:
Thank You so much!

Bodmer, 2021-03-27:
I have updated the master copy on github whith the changes I think are needed. please test.

Only the files [TFT_eSPI_ESP32.h and TFT_eSPI_ESP32.c have been updated ](https://github.com/Bodmer/TFT_eSPI/tree/master/Processors)so you do not need to copy the entire library over.

flykarlos, 2021-03-27:
Works beautifully now! outstanding, thanks!

Bodmer, 2021-03-27:
Yay! I think you are the first user of that feature.

Enjoy!

flykarlos, 2021-03-28:
I will test it extensively, thank You so much!

dinas80, 2021-03-28:
I have the same tft model as "drewp" and I want to make it work with esp32 so can anyone please tell me how to wire it, and what code did you use or which library settings

drewp, 2021-03-29:
Hi, I'm happy to try to help, but note that this is a ticket for the TFT_eSPI which I am not using. If you want to use my esphome code, we should discuss that on the esphome discord or in an esphome issue.

Here are my wiring notes:
```
# ili9486 setup
# http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO
# (https://www.aliexpress.com/item/4000055026873.html $12)
# and https://circuits4you.com/wp-content/uploads/2018/12/ESP32-Pinout.jpg

#       GND      - GND  Power ground
#       5V       - 5V   5V power input
#       NC       - 3V3  3.3V power input, this pin can be disconnected
#       GND      - LCD_RST      LCD bus reset signal, low level reset
#       GND      - RESET      (not sure- just tied it to gnd)
#       NC         - F_CS         (not sure)
#
#       GPIO16   - LCD_CS       LCD bus chip select signal, low level enable
#
#       GPIO17   - LCD_RS       (aka DC) LCD bus command / data selection signal, low level: command, high level: data
#       GPIO5    - LCD_WR       LCD bus write signal
#       GPIO18   - LCD_RD       LCD bus read signal

#       GPIO33   - LCD_D0       LCD 8-bit data Bit0
#       GPIO32   - LCD_D1       LCD 8-bit data Bit1
#       GPIO13   - LCD_D2       LCD 8-bit data Bit2
#       GPIO12   - LCD_D3       LCD 8-bit data Bit3
#       GPIO14   - LCD_D4       LCD 8-bit data Bit4
#       GPIO27   - LCD_D5       LCD 8-bit data Bit5
#       GPIO26   - LCD_D6       LCD 8-bit data Bit6
#       GPIO25   - LCD_D7       LCD 8-bit data Bit7

```
Then I hacked this esphome fork (of someone else's fork) together: https://github.com/drewp/esphome/commit/c160835738d36949bec7b70ec860ad01cb8ea5e4

I receive mqtt lines and display them with this esphome code: https://github.com/drewp/homeauto/commit/02c7c33bdc6c80afb11735cd6e98f3740f441ecd

Bodmer, 2021-03-29:
Note that the display used by the OP has NOT got a RD pin, see image posted, so @drewp must have a different display.

Check the pinout of the display to see which it is. Post a link or picture of the back of the board to be sure.

dinas80, 2021-03-29:
@drewp Thank you so much, I really appreciate your help
I'm not familiar yet with esphome so it'll be very hard to keep up with you, once I'm if there were any issues I'll get back to you.

dinas80, 2021-03-29:
@Bodmer there's my board 
I got to display, but I'm facing a problem with the touch demo, and when I tried your solution for issue #117 it says can't compile so if you can help me with this it'll be great 
[image]

flykarlos, 2021-03-29:
Hi Bodmer, can You advice if antialiased/smooth fonts may be used with good effect with pseudo_16_bit mode? I tried some examples, but font edges are ragged..

Bodmer, 2021-03-29:
@dinas80 Your display will have a resistance based analogue interface (assuming the screen has this fitted!).  The TFT_eSPI library does not provide support for that type of interface. The complication is that the touch interface shares the pins with other TFT screen interface pins!

The good news is that a user has created a library for these UNO style displays.  See last paragraph of[ this section](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support).

Bodmer, 2021-03-29:
@flykarlos Good point, you will not be able to use the smooth fonts as the anti-aliasing will not work.

zaydsaoudi, 2022-04-06:
Hi Bodmer , i need some help with tft shield uno 2.8 
I have touche problem with parallel interfacing , i use the tft_espi and touche.h library , i tried touch_cs too 
Nothing working with me

Haimana, 2024-11-18:
Hello,

I'm trying to connect a 3.5 TFT display, like the one from the first post to a Nodemcu-32s, using 8bit parallel wiring.
I also need to have the wifi functional and I use different GPIOs.
Because I use Platformio/VS Code, I tried to use the platformio.ini for configuring my display, and not to edit the library's User_Setup.h file.

My platformio.ini looks like this:

;PlatformIO User notes:

;It is possible to load settings from the calling program rather than modifying
;the library for each project by modifying the "platformio.ini" file.

```
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
lib_deps = bodmer/TFT_eSPI

build_flags =
	-D USER_SETUP_LOADED
	-D ESP32_PARALLEL
	-D ILI9486_DRIVER
	-D PSEUDO_16_BIT
	-D TFT_RD=-1  	;RD pin not used, set to -1
	-D TFT_WR=4  	;Write strobe pin
	-D TFT_DC=5  	;Data/command pin
	-D TFT_CS=21  	;Chip select pin
	-D TFT_RST=22	;Reset pin

	-D TFT_D0=12
	-D TFT_D1=13 
	-D TFT_D2=14
	-D TFT_D3=15 
	-D TFT_D4=16
	-D TFT_D5=17
	-D TFT_D6=18
	-D TFT_D7=19

	-D LOAD_GLCD=1
	-D LOAD_FONT2=1
	-D LOAD_FONT4=1
	-D LOAD_FONT6=1
	-D LOAD_FONT7=1
	-D LOAD_FONT8=1
	-D LOAD_GFXFF=1
	-D SMOOTH_FONT=1
	-D SPI_FREQUENCY=27000000

```

But my display have a lot of glitches and after a while turns full white.
I made the connections like this:

```
Connect D0 to D8
Connect D1 to D9
Connect D2 to D10
Connect D3 to D11
Connect D4 to D12
Connect D5 to D13
Connect D6 to D14
Connect D7 to D15

```

Obviously I'm doing something wrong or I forgot something because first time, when I have the User_setup file edited, the display was OK but my modifications was lost, rewritten by platformio at some "wrong move" of mine. So I am not able to compare what I did there and it worked and is not done in this setup.
If need it, I can upload some pics, but for now I'm trying to keep the topic as clean as possible

Thank you for your time and help
BR, 
Seb

newmarouane, 2025-01-18:
@drewp 
hello dear @drewp ,
I hope you are fine. Please I need your help. I have an ILI9488 parallel screen and I want to connect it to an esp32-wroom-32u via esphome.
Can you help me please?
Thank you

drewp, 2025-01-23:
@newmarouane It's so much more efficient if you explain how far you got and what went wong. 

But, as I wrote above, this thread is on the TFT_eSPI project. You should probably start at https://esphome.io/components/display/ili9xxx.html or  https://github.com/esphome/issues/issues?q=is%3Aissue%20state%3Aopen%20ili9488

newmarouane, 2025-01-30:
> Mine had to be held at RD=high for communication to work. HTH
> 
> I'm up and running in esphome now! [image]

hello @drewp ,
I connected an 3.5" parallel touchscreen to my esp32-wroom-32u board via esphome. I arrived to display texts on it using "it" lambda commands, but when I use lvgl component, it shows a black screen only even if I reduce buffer_size to 19% or less.
Can you help me please to solve this problem?
Best regards