Bodmer/TFT_eSPI #300: Question about fonts!

marslolx, 2019-02-06:
Hi Bodmer, I know it must be silly but how are new sources included in the library? What would be the files that must be modified to include them? I was trying but it seems that I lack practice in this since I can not make it work.
Actually my problem is because none includes the letter "ñ". Do you know any that include it?
Thank you very much already!

Petros144, 2019-02-06:
Hi Marslolx,

I had the same problem with the German letters "ÄÜÖ äüö"

Making custom fonts is the key to succsess.
(there are 100 other ways making these, but i will tell you mine)

I used a Tool to convert normal windows fonts to Custom TFT fonts:

Here is the Youtube Video that shows how exactly you can do this

[https://www.youtube.com/watch?v=L8MmTISmwZ8](url)

If thats done, include this in the Custom fonts files and your sketch, and it should work!

hope this helps.

marslolx, 2019-02-06:
Thanks a lot, Petros144! I'm going to try your way to see if I succeed and tell you!

Petros144, 2019-02-07:
in addition to the Link above,

if the font wont convert right or makes wired stuff, you can check your font data in this online viewer and edit the "wrong" chars, im not shure why this happens but its nice to be able to edit the font.

Link to the Editor:

https://tchapi.github.io/Adafruit-GFX-Font-Customiser/

and some tips from the makers:

https://www.foobarflies.io/adafruit-gfx-font-customiser/

it took me ages to get my own perfect font, hopefulle you can do this faster.

____

Petros

Bodmer, 2019-02-08:
I have just added support to the TFT_eSPI library for the extended character set 32-255 in an Adafruit_GFX compatible font.  A [new example has been added here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/Local_Custom_Fonts).

The font file is attached to the sketch, the file was generated by using the same process as detailed in this [YouTube link](https://www.youtube.com/watch?v=L8MmTISmwZ8) kindly above posted by @marslolx.

closed 2019-02-08 by Bodmer

Bodmer, 2019-02-08:
Screenshot from example:
https://i.imgur.com/tGh8rzv.png

Bodmer, 2019-02-08:
[image]