Bodmer/TFT_eSPI #302: Problem with self made .vlw smooth fonts

plueschpruem, 2019-02-09:
Great library!
But n
[ComSa20.vlw.zip](https://github.com/Bodmer/TFT_eSPI/files/2847811/ComSa20.vlw.zip)

o matter what font I generate, it always ends up with being displayed at about 2x the line height it should have.
As a result drawString() "erases" the 1st line when a 2nd line is drawn.
Adjusting gFont.yAdvance helps with tft.println(), but drawString() is still broken.

I used the latest Processing3.app on OSX and the fonts behave as expected in Processing (not an expert on that app, but printing several lines looks normal).

Find attached one sample I generated.
The fonts included with the library behave normally, so I am kinda out of ideas…

Bodmer, 2019-02-10:
The problem is that the font character 0x0080 in the file is an invisible code but the font file gives it a height of 40 pixels. All other characters have a maximum height of 17 pixels. 

Try eliminating that character from the font by missing out that character.

For example:

```
  0x0021, 0x007E, //Basic Latin, 128, 128, Latin (52 characters), Common (76 characters)
  0x0081, 0x00FF, //Latin-1 Supplement, 128, 128, Latin (64 characters), Common (64 characters)
```
Let me know if that works.

plueschpruem, 2019-02-10:
Thanks for the swift reply!

I tried the processing sketch again and changed the range for Latin-1 Supplement to 0x00A1 as suggested.
instead of 222 chars it returned 189
**And it works now! Thank you so much!**
Even saved roughly 2kB in size too.

The kicker is, that font creation is speed by a factor of 5x now (from RUN to displaying the font), the program really seems to chew on those empty chars for a long time.

I changed the sketch to reflect this issue.
I also added some lines to output the font list to a file, since the console purged everything above 690 for me (I got a lot of fonts installed).

[create_font_tft_eSPI.pde.zip](https://github.com/Bodmer/TFT_eSPI/files/2848869/create_font_tft_eSPI.pde.zip)

closed 2019-02-10 by plueschpruem

Bodmer, 2019-02-10:
The character range 0x0080 - 0x00A0 created characters containing 1 transparent pixel according to the metrics but with a declared width to 20 and height of 40. This plus the slow speed you noted suggests a bug in the Processing IDE. When TFT_eSPI loads the font it looks for the tallest character ascent plus descent to determine the line spacing, hence the large line gaps which are to avoid line overlap.

I had a look and the font processing code in the Processing IDE appears to have been updated recently so maybe it is falling over on some valid characters? I will load the latest version to try. The other possibility is a bug in the original ttf font file.

What is the name of the font file you are using?

Thanks for the updated sketch. I included instructions in the sketch to increase the console window line count to 1000 but I think your approach of spooling the list to a file is much easier from a users perspective and caters for even longer font lists.

Bodmer, 2019-02-10:
OK, I see you used ComicSansMS.ttf

plueschpruem, 2019-02-10:
Yeah I used the most evil font possible :-)
But I had the problem with EVERY font I tried.
Processing is the latest OSX version.
Maybe my old OSX contributes to the situation - this Mac is still on 10.11…

Bodmer, 2019-02-10:
I have updated the library to rely on the font line spacing metrics produced by the Processing IDE. 

Derivation of the line spacing from the metrics of each font glyph has proved to be unreliable due to the way non-existent characters are incorporated into the font file by the Processing IDE. For non-existent characters the Processing IDE appears to default to the ~1X point size for width and ~2x the point size for height, this then produces disproportionately large line spacing.

I have added in your code for the font listing to file.

Thanks for reporting this issue and also for providing the enhancement to the font generator sketch.