Bodmer/TFT_eSPI #303: drawString, touch and so on

RA-NGR, 2019-02-10:
Hi, Bodmer. I've just updated to last snapshot and i've a lot of problems.
Test sketch is
```#include <SPI.h>
#include <TFT_eSPI.h>       // Hardware-specific library

#define AA_FONT_SMALL  "ArialRMTP15"

TFT_eSPI tft = TFT_eSPI();

void setup() 
{
  Serial.begin(115200);
  SPIFFS.begin();
  tft.begin();
  tft.setRotation(0);
  tft.fillScreen(TFT_BLACK);
  tft.loadFont(AA_FONT_SMALL);
  tft.setTextWrap(true);
  tft.setTextDatum(BC_DATUM);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);  
  tft.drawString("6°/-1°", 35, 252);
}

void loop()
{
}
```
Result is
[image]
Serisl output is
[image]
When i've comment out strings 4601-4606 the problem is gone. 
The second issue is touch isn't working at all. With previous version of your library the touch working well.
I using 320x240 TFT based on ILI9341/XPT2046 driver and Wemos D1 mini with such connections: 
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V
#define TOUCH_CS PIN_D0 //PIN_D2     // Chip select pin (T_CS) of touch screen

Bodmer, 2019-02-10:
Thanks for reporting this.

I see I left some temporary debug code in the drawString() function, I have removed that now.

I will investigate the touch screen issue.

RA-NGR, 2019-02-10:
Thank you. will wait

Bodmer, 2019-02-10:
Yes, you are right the sumX values were wrong. This is now corrected.

Hopefully the only issue remaining is the Touch screen handler. I will run some tests on an ESP8266 board. I think I only tested the last update on and ESP32 so maybe it is an ESP8266 only issue. Hopefully I will figure this out soon.

Bodmer, 2019-02-11:
I have run a test on the touch controller function and it is broken at the moment.

The "Test_Touch_Controller" ouptu works OK, but as soon as a single pixel is written to the TFT the touch controller stops working.

I will investigate further and try to identify the root cause.

RA-NGR, 2019-02-12:
Good news, will wait for corrected code )

achillhasler, 2019-02-12:
Have a try with https://github.com/achillhasler/TFT_eTouch. If Touch in TFT_sSPI not working.

RA-NGR, 2019-02-12:
Thanks, but additional class will reduce available RAM, so I,ll wait for Bodmer code ))

Bodmer, 2019-02-14:
The touch functions in the library are now back to a working condition. Repaired version is 1.4.5.

Post back if this does not work for you.

closed 2019-02-14 by Bodmer