Bodmer/TFT_eSPI #304: Does the library support ILI9481 SPI mode

aruldd, 2019-02-11:
Hi,

I just wanted to know if the library supports ILI9481 in SPI mode. I have found one setup configuration for ILI9481 in parallel but not in SPI.

Thank you in advance.

Bodmer, 2019-02-11:
Yes, it does support the SPI ILI9481.  I will see if I have an option file.

It is just a case of picking one of the other config files and adapting it.

Bodmer, 2019-02-11:
Setup 20/21 should work with ESP8266/ESP32:

#define ILI9488_DRIVER

changed to:

#define ILI9481_DRIVER

Bodmer, 2019-02-13:
If this does not work then post back and I will investigate.

closed 2019-02-13 by Bodmer

aruldd, 2019-02-13:
It doesn't work but I am not sure about the connections, I am using the SPI pins on the 40 pin connector of the LCD itself which has an ILI9481 driver.  [This](https://www.aliexpress.com/item/2-4-2-8-3-2-3-5-inch-TFT-LCD-display-screen-touch-panel-40/32837085500.html?spm=a2g0s.9042311.0.0.18914c4dnnNfUn) is the display.

aruldd, 2019-02-13:
I am unable to get it to work, I have followed the pinout from the manufacturer page. Can you point me to the right schematic/connection if that's not too much to ask.

Thank you!

Bodmer, 2019-02-13:
List all the power, configurarion (IMxx) and all pin conections from the ESP??? to display you think are right and I will take a look.

aruldd, 2019-02-14:
I have listed all the pins in the attached document.
[lcd pin connections.pdf](https://github.com/Bodmer/TFT_eSPI/files/2863547/lcd.pin.connections.pdf)

Bodmer, 2019-02-14:
Looks OK apart from config pins 38, 39, 40. According to the vendors web page (from link you provided) these pins should be: 38=GND, 39=3.3V, 40=3.3V.

aruldd, 2019-02-14:
I am trying that right now, But isn't that for Parallel 8 bit mode. I am connecting the display directly to ESP32 without any shift-register setup like in Rpi displays.

Bodmer, 2019-02-14:
I based the pin config on the table provided on the vendors page, but that may be wrong (and be for the ILI9341?) as it is different to the data sheet. One possibility is that you actually have a ILI9488 display, as you cannot always believe what some Chinese vendors say on their web page!

aruldd, 2019-02-16:
I tried everything but nothing seems to work. The driver is printed on the LCD itself as 9481. I am giving up on this for now. Is there any IPS LCD with the same resolution which will work out of the box with this library?

Bodmer, 2019-02-16:
The simplest and most performant solution is an [8 bit parallel type display](https://www.ebay.co.uk/itm/3-5inch-TFT-Color-Screen-Module-320X480-Ultra-HD-LCD-Shield-for-Arduino-ILI9481/323525273841?hash=item4b539be8f1:g:l2AAAOSwxyJaw0T8:rk:1:pf:1&frcectupt=true) with an UNO format ESP32 board with the mods detailed in the library ReadMe.

For SPI (4-wire interface needed) the most reliable/quality supplier for sending you what they actually advertise is [this Buy Display one ](https://www.buydisplay.com/default/lcd-3-5-inch-320x480-tft-display-module-optl-touch-screen-w-breakout-board) which uses the ILI9488 controller.  Performance is lower as it needs 3 bytes per pixel instead of 2 for the ILI9481.  I have one without the touch screen fitted and the font chip is not needed.

SPI displays based on the ILI9481 pop up on ebay occasionally but it is a lottery which controller you actually get.

The lower resolution ILI9341 displays are a good beginners display as they are so common and so many libraries and Arduino boards support them, hence the library/processor choice is better in case of difficulties.

yo2ldk, 2022-02-26:
I also buy it an SPI TFT IPS ILI9481 display, but it not work at all with ESP32, in library I not found driver only for parallel
 use different driver like for 9488, 9341 and so on (all for esp32 from user setup select) all I have is black screen with some small dark points like on old tv's without signal
 any chance to make a driver for this type?
  Thank you in advance!

ParallelTwin-Rider, 2022-03-25:
I think I found a solution on this histrical issue.   The data sheet of Ali Express are wrong amonst shops who sell ILI9481.
I spent whole day and finally found true data sheet.

To use ILI9481 40pin model as 4 wire SPI, followings are the right pin assignment of ILI9481 40p model.  Also display refresh parameter should be 10MHz as Bodmer mentioned before.

Pin 1-4 : touch panel 
Pin 5 GND
Pin 6, 7  VCC (3.3V)
Pin 8 (FMARK) : no use as far as using multiple syncronized display, may be
Pin 9 :  SPI CS
Pin 10 :  SPI DC
Pin 11 (WR/SPI_A0 4 Wire): SPI SCLK
Pin 12(RD) :  VCC (need to connect to VCC to use as SPI)
Pin 13:  SPI MOSI
Pin 14:  SPI MISO
Pin 15:  Reset
Pin 16:  GND
Pin 17-32  GND
Pin 33   LED Anode (3.0-3.4V) = 3.3V
Pin 34-36  LED Cathode = GND
Pin 37 GND
Pin 38-40  VCC to use this as SPI

That's it !!

Bodmer, 2022-03-25:
Thanks for that. So, it looks like the manufacturers information had the signals on pins 10 and 11 swapped. No wonder @aruldd could not get it to work. Well done for finding that!

ParallelTwin-Rider, 2022-03-26:
One memo in addition to the comment above.  If the form factor of ILI9481 is 40pin flexible wire, many of purchaser might buy 40p adopter as well.   The printed Pin number in front and back is not identical.  So the user should align with the side where the 40pin socket is placed.  This could provide confusion too.

Bodmer, 2022-03-26:
Yes, I have and FPC board that swaps the pins. The ribbon does not make contact the other way around! It depends whether the FPC connector has top or bottom contacts and the orientation of pins on the TFT ribbon:
[image]

ismaelit, 2022-08-15:
> Yes, I have an FPC breakout board that swaps the pins. 

I have the same breakout board. So the FPC contacts are only on the bottom (board) side, right? The number 1-40 printed on the flat are correct?

ismaelit, 2022-08-20:
> Pin 1-4 : touch panel Pin 5 GND Pin 6, 7 VCC (3.3V) Pin 8 (FMARK) : no use as far as using multiple syncronized display, may be Pin 9 : SPI CS Pin 10 : SPI DC Pin 11 (WR/SPI_A0 4 Wire): SPI SCLK Pin [...]

Hello friend. I have made exactly this (plus a lot of other tries), I have backlight turning on, but black screen. 

On User_Setup.h I have:

#define ILI9481_DRIVER

#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5  // Chip select control pin
#define TFT_DC   26  // Data Command control pin
#define TFT_RST  -1  // 3.3v, also tried a esp32 port and nothing

#define SPI_FREQUENCY  10000000

[image]

I think I could have damaged this display, but never used more than 3.3v on any pin.