Bodmer/TFT_eSPI #305: Can M5Stack createSprite()   more than 238x238?

TridentTD, 2019-02-11:


Bodmer, 2019-02-11:
Not unless you have the M5Stack Fire which has extra PSRAM, then you can probable create 10 Sprites of 480x320 pixels!

Alternatively use an 8 bit Sprite or partial screen Sprite and say two screen update cycles for different screen areas.  Writing to a Sprite is fast so clearing, redrawing and pushing elsewhere on the screen may be practical for you. The library crops graphics in a Sprite so you can have a full screen drawing function with an offset of say 0,0 for the top half of the screen and 0, -160 for the bottom half.

TridentTD, 2019-02-12:
If  I use the M5Stack Fire , 
how can i declare createSprite() for using the extra PSRAM ?
It's automatic for  PSRAM by default ?

Bodmer, 2019-02-12:
You have to enable the PSRAM in the Arduino  IDE menu then it is automatic.

closed 2019-02-12 by Bodmer