Bodmer/TFT_eSPI #306: please Help !   st7789 240x240 inverted colors

VikJr, 2019-02-12:
please Help !   st7789 240x240 inverted colors. Black -> White , .....
--------
user setup.h

#define ST7789_DRIVER // Define the screen size below for this display
 #define TFT_SDA_READ

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue - If uncommented - nothing will change.
//#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red - If uncommented - nothing will change.

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TFT_BL   32  // LED back-light (only for ST7789 with backlight control pin)
-------

Bodmer, 2019-02-12:
Try:

```
invertDisplay(boolean i)
```
add it after tft.init(); in the setup() section of your sketch.

VikJr, 2019-02-12:
Today I updated the library and found the line 
 #define TFT_INVERSION_ON
in the file   "User_Setup.h"
now everything works correctly!

Thank you so much for your excellent work and fast support!
I have never seen the code so detailed and clearly described.

closed 2019-02-12 by VikJr

Bodmer, 2019-02-13:
Thanks for the feedback.  Yes, you found the better solution, I should have suggested that as then any sketch should run without modification.

The detailed comments is because most folk don't like reading manuals and also writing support documents tends to stay at the bottom of my "to do" list!