Bodmer/TFT_eSPI #307: ..i tried ILI9335 on R32 but return error upload

SurenBono, 2019-02-14:
...the test sketch MCUFRIEND_KBV, ili9335 TFT  shield worked on UNO...but why is it not on esp32 r32 uno style... even after hard solder instruction were applied..? ...
[image]

prenticedavid, 2019-02-14:
You seem to have opened an issue.   Edited it. Closed it.
Then re-opened.

The original message seems to show a controller with ID = 0x9335 running an example sketch. 
You must have done the hardware mod on the TTGO R32 board correctly.

What is your question?

David.

Bodmer, 2019-02-14:
Looks like this has been posted in the wrong place.

As David say's, the TFT_eSPI library does not support ILI9335 displays.

closed 2019-02-14 by Bodmer