Bodmer/TFT_eSPI #308: [feature request] Overflow flag for tft.println()

plueschpruem, 2019-02-14:
I use `tft.println()` with `tft.setTextWrap(true, false)` to print strings of unknown length on the top half of the screen, then check `getCursorY()` if I went over 1/2 - if not I add a picture, otherwise I leave the rest blank.

This only works until the cursor overshoots the screen's Y resolution.
From _Smooth_font.cpp:_ `if (cursor_y >= _height) cursor_y = 0;`
Maybe we can add a flag here that tells if the last operation overshot Y?

I quickly did that for my project and it works:

Declaration in _Smooth_font.h:_
`virtual bool getOvershot();`

In _Smooth_font.cpp_ I added:
```bool cursorYovershot = false;
bool TFT_eSPI::getOvershot() {
	if (cursorYovershot == true) {
		cursorYovershot == false;
		return true;
	} else {
		return false;
	}
}
```
And edited drawGlyph():
 ```
 […]
    if (code == '\n') {
      cursor_x = 0;
      cursor_y += gFont.yAdvance;
      if (cursor_y >= _height) { 
	    cursor_y = 0;
      	cursorYovershot = true;
	  } else {
      	cursorYovershot = false;
	  }
      return;
    }
[…]
```
Maybe you could adapt this in some form?
Feel free to code it properly ;-)

Bodmer, 2019-02-14:
Thanks for the suggestion. A flag to indicate a wrap has occured on the X axis might also prove useful.

I will think on this further as there are also other potential status or error flags that could be useful.

plueschpruem, 2019-02-15:
Another great feature would be an option to have fonts be true monospaced.
Most monospaced fonts I tried still have too thin spaces and punctuation. I guess the original TTF / OTF ones use kerning tables to compensate.
Maybe there is a simple way - like centering each glyph to a evenly distributed space the width of the widest glyph of the font.
I also noticed that some handwriting fonts overlap partially- which kinda looks great tho.

plueschpruem, 2019-02-18:
Just noted that every time I issue loadFont() the current font will be unloaded and the requested one loaded, even if the new one is the same as the old one.
Would be cool to have a check in there to prevent useless reloading and circumvent checking yourself with if (tft._gFontFilename != "/" + String(fontName) + ".vlw")

closed 2020-12-16 by Bodmer