Bodmer/TFT_eSPI #309: Weird positioning

themakerbro, 2019-02-17:
Thanks for the port to NodeMCU. I just tried it and get a weird wrap-around for the default sketch. Any suggestions?

[image]

Bodmer, 2019-02-18:
The skew is due to a change in the setAddrWindow parameter. I have updated the Github example.

The offset is due to the display driver having a pixel offset. Try different rotations, you should find one rotation that has no offset.

The offset happens because the controller can drive a display with more pixels, so it is dependant on the way the manufacturer wires up the pixels.

themakerbro, 2019-02-18:
Ah, I see. I hadn't gotten a chance to dig into the code and you already solved it. Thanks so much!

[image]

closed 2019-02-18 by themakerbro

Bodmer, 2019-02-18:
Great.
Note that I never got around to making the sketch support 2 eyes...