Bodmer/TFT_eSPI #310: Problems with filled shapes and fillScreen

gilshahar7, 2019-02-18:
Hi, i am fairly new to the ESP8266 so this might be a stupid mistake on my side.

I have finally got the RaspberryPi ILI9684 LCD (480X320 with touch) working on my NodeMCU (ESP8266).

It is showing the examples but there seem to be an issue with fillScreen and any other filledShape kind of methods.

Here are a few pictures, 

This is using `tft.fillCircle(320/2, 480/2, 10, `0x0000);`
[image]

Then `tft.fillCircle(320/2, 480/2, 20, 0x0000);`
[image]

And this is `tft.fillCircle(320/2, 480/2, 100, 0x0000);`
[image]

This is what i get when i use `tft.fillScreen(0x0000);` 
Notice the marked corner, there is a small line with black color, the rest of the screen always gets colored white no matter what color i try
[image]

This is the full code that i used in the photos: 

```#include <SPI.h>

#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

#define TFT_GREY 0x5AEB // New colour

void setup(void) {
  tft.init();
  tft.setRotation(0);
}

void loop() {
  

  tft.fillScreen(0x0000);

     // tft.fillCircle(320/2, 480/2, 100, 0x0000);

  tft.setCursor(50, 50, 2);

  tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);

  tft.println("Hello World!");

  delay(10000);
}

```

My User_Setup.h has `#define RPI_ILI9486_DRIVER` uncommented and my User_Setup_Select.h has `#include <User_Setups/Setup10_RPi_touch_ILI9486.h>` uncommented.

I think my screen wiring is fine but here it is anyways.

Using this diagram: 
[image]

I connected the folowing:

```
Display SDO/MISO  to NodeMCU pin D6
Display +5V       to NodeMCU pin VIN
Display SCK       to NodeMCU pin D5
Display SDI/MOSI  to NodeMCU pin D7
Display DC (or AO)to NodeMCU pin D3
Display RESET     to NodeMCU pin D4
Display CS        to NodeMCU pin D8
Display T_CS        to NodeMCU pin D1
Display GND       to NodeMCU pin GND (0V)
another Display GND       to NodeMCU pin GND (0V)
```

Touch worked fine.

Hopefully i didn't forget to give any important information. Thanks for your time and effort

armageddon421, 2019-02-18:
Have the exact same problem, been looking into it the whole evening already. Funny thing ist, everything worked fine just yesterday.

armageddon421, 2019-02-18:
Seems to be related to fastvline. Rainbow example has the same issue. Probably something ist wrong with writepattern.
@gilshahar7 can you try disconnecting miso from the display and shorting it to mosi? Seems like this helps, but makes the touchscreen unusable.
Must be a problem in writepattern, so the miso is not ignored and the registers are filled with whatever is on the miso line, so only the first set of registers is properly sent.

gilshahar7, 2019-02-18:
> Seems to be related to fastvline. Rainbow example has the same issue. Probably something ist wrong with writepattern.
> @gilshahar7 can you try disconnecting miso from the display and shorting it to [...]

Thanks for the quick reply! can you explain more about what "shorting it to mosi" means? should i have a wire going from display's mosi to display's miso?

armageddon421, 2019-02-18:
Sorry for the sloppy text, was only on my phone.
- Disconnect MISO from display
- Connect to MOSI instead
So you have a wire going from the Display's MOSI to both MOSI and MISO on the ESP side.

armageddon421, 2019-02-18:
This is what the rainbow example looks like in this situation.

[image]

gilshahar7, 2019-02-18:
> Sorry for the sloppy text, was only on my phone.
> 
> * Disconnect MISO from display
> * Connect to MOSI instead
>   So you have a wire going from the Display's MOSI to both MOSI and MISO on the ESP side.

Yes this fixed it for me! basically i have Display's MOSI connected to D6 and D7 at the same time.

As of now i dont need touch but it seems like an issue that can be fixed by software so ill leave it to the experts.

Thanks :)

armageddon421, 2019-02-18:
I'm glad I could help. It isn't fixed for me, though, as I need the touchscreen to work. I will continue trying to find out what this is about, but I kind of reached the limit of what I can do.
Just yesterday, everything worked fine, and I just can't find out what changed. To rule out a broken display or esp8266, I swapped out displays and ESPs, and I tried with Arduino IDE and Platformio.

armageddon421, 2019-02-18:
Problem persists in Arduino IDE with esp8266 core versions 1.30, 1.4.2 and 1.5.0. Tested with TFT_eSPI version 1.4.5 (latest in Arduino IDE)

armageddon421, 2019-02-18:
Also, using all the latest versions in PlatformIO causes the same problem.
esp8266 core should be 2.5.0 in that scenario.

`[env:d1_mini]
platform = https://github.com/platformio/platform-espressif8266.git
board = d1_mini
framework = arduino

lib_deps = https://github.com/Bodmer/TFT_eSPI.git

build_flags =
  -Os
 ; -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  -DRPI_ILI9486_DRIVER=1
;  -DTFT_WIDTH=320
;  -DTFT_HEIGHT=480
;  -DTFT_MISO=12
;  -DTFT_MOSI=13
;  -DTFT_SCLK=14
  -DTFT_CS=15
  -DTFT_DC=0
  -DTFT_RST=2
  -DTFT_BL=16
  -DTOUCH_CS=5
  -DLOAD_GLCD=1
  -DSPI_FREQUENCY=20000000
  -DSPI_TOUCH_FREQUENCY=2500000
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1
  -DLOAD_FONT6=1
  -DLOAD_FONT7=1
  -DLOAD_FONT8=1
  -DLOAD_GFXFF=1
  -DSMOOTH_FONT=1`

Bodmer, 2019-02-19:
Hi,
Thanks for reporting this.

The pictures indicate that up to 32 correct pixels are being drawn and the SPI output buffer is then getting flushed (with 0xFFFF = white) because it is in read+write mode and it should be in write mode only. Connecting MOSI to MISO re-circulates the transmitted data back into the  SPI buffer, so appears to fix it.

The writePattern() function in the Arduino ESP8266 core SPI library has a feature (aka bug), it changes the SPI register flag bits but then fails to put them back to the state they were in before the call. This has manifested itself due to recent changes in the TFP_eSPI library.

I have [added a line to the library](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L2792) that should fix it.  Please report back if this worked for you (or not).

armageddon421, 2019-02-19:
That fixed it. Thank you very much!
I must've been close, but couldn't quite put my finger on it.

gilshahar7, 2019-02-19:
Yes added the line and it works with the normal wiring of the display. thanks a lot Bodmer, i appreciate your work.

Bodmer, 2019-02-19:
Great, glad to hear it is working now

closed 2019-02-19 by Bodmer