Bodmer/TFT_eSPI #311: Sprite can't display smooth font correctly.

TridentTD, 2019-02-19:


TridentTD, 2019-02-19:
When I use TFT_eSPI version 1.3.8.
the sprite can display smooth fonts correctly.

<p align="center">
  [image]
</p>

Bodmer, 2019-02-19:
This looks like an SPI interface timing issue. What SPI rate are you using? Does it happen at a lower clock rate?

TridentTD, 2019-02-19:
TFT_eSPI version 1.4.6(master)
all of these configs  still  display incorrectly.

-------------------------------------------------------
#define SPI_FREQUENCY   1000000
#define SPI_FREQUENCY   5000000
#define SPI_FREQUENCY  10000000
#define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000 // Actually sets it to 26.67MHz = 80/3
#define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS  
#define SPI_FREQUENCY  80000000
----------------------------------------------------

Bodmer, 2019-02-20:
Thanks for reporting this and for providing an example that demonstrates the problem.

This was a bug in the Sprite drawPixel function for the bounds check.  It is now fixed in the master.

closed 2019-02-20 by Bodmer

TridentTD, 2019-02-20:
TFT_eSPI version 1.4.7(master) is OK.

Thank you very much.