Bodmer/TFT_eSPI #312: How do I use exotic unicodes?

plueschpruem, 2019-02-19:
I'm trying to use some symbols from this font:
fontawesome-free-5.7.2-web from https://fontawesome.com/download

Processing generates and displays it correctly, but showFont() only display the first two chars and then blank pages.

But no matter what I do, I can't print or draw any of them myself.
Example from above font would be code 0xF029.

Bodmer, 2019-02-20:
Post a simple example sketch that demonstrates the problem, complete with the required vlw font file. Keep the example small and the font file with as few characters as demonstrates the problem.

Do not include characters that do not exist in the font. (These show as boxes normally in the Processing window)

plueschpruem, 2019-02-20:
See attached sketch...
It shows both fonts correctly in showFont() now (my symbols were @ 160px, here resized to rule out positioning/RAM errors), but masking the unicodes is still a mystery to me :-)

I know "regular" stuff like ® or Ö can just be written as such in code.
But how to mask the exotic ones with code in a String?
"\uF026" / "\\uF026" / "\xF026" / "\\xF026"…?

[unicode Test 2.zip](https://github.com/Bodmer/TFT_eSPI/files/2884507/unicode.Test.2.zip)

Bodmer, 2019-02-20:
This is the format to use:
```
tft.print("\uF026");
```
Always use the full 16 bit hex value. You can mix text and codes and string codes together, for example:

```
tft.print("\uF026Hello");
tft.print("\u00F0\uF026");
```

closed 2019-02-20 by Bodmer

plueschpruem, 2019-02-20:
Aaaah! With the print() functions, not draw().
Thanks a lot once again!

The 160 px size still craps out tho, the first two chars show fine, #3 to #end are a no-show…
I guess I will have to use .jpegs here…

Bodmer, 2019-02-20:
Font characters have a 127 pixel maximum height = 127 maximum for point size.  This is to reduce the memory needs for the font metrics but using 8 bit signed ints.  The size limit depends on the position of the baseline in the font, so 127 is actually the limit for the character height above the baseline.

plueschpruem, 2019-02-20:
Yet another AAAAAHHHH!
So I most def have to resort to JPG.
Thank!s

Bodmer, 2019-02-20:
I had a look at where this limitation arises and have increased the maximum permitted font size.

160 pixels should work now (but I have not tested it). Updated files are Smooth_font.h and Smooth_font.cpp in the Extensions folder.

plueschpruem, 2019-02-21:
Updated and it works flawlessly.
Only problem now is positioning them dead center because print() statement don't support TextDatum…
But manual positioning will do for now.

Bodmer, 2019-02-22:
drawString() was supposed to handle UTF-8 encoded strings but there was a bug in the line buffer decoder...

This has now been fixed so this (for example) will now work:
```
  tft.setTextDatum(MC_DATUM);
  tft.drawString("\uF026", 80, 80);
```

Bodmer, 2019-02-22:
You can also mix the unicode with text if the font contains all the needed characters, e.g.

```
  tft.setTextDatum(MC_DATUM);
  tft.drawString("Speaker \uF026", 80, 80);
```